cncjs/cncjs #325: Overlaying machine travel on to visualiser

TobyGWilliams, 2018-05-11:
I'd like to have a way of overlaying the machine travel on to the visualizer to see if my gcode and work coordinates will cause the machine to run into it's limits. 

And I'd like to have a go at building it myself despite having no experience of Three.js or React (I'm an angular and D3.js guy so far).

Maybe somebody will build it properly for me if I can cobble it together.

So far I've got:

[image]

**So far I've:**
- Added MachineTravel.js to the Visualiser folder
- Imported MachineTravel.js into Visualiser.jsx
- Imported the Workspace json from store
- Drawn the appropriate rectangle

**Left to do:**
- Figure out how to offset the rectangle, so that as the work offsets change, the rectangle stays aligned to the machine co-ordinates not the work offsets. I think this can be calculated by taking the difference between the Machine Co-ordinates and Work Co-ordinates.
- Add some machine configuration UI to use to maintain the axis limits.
- Extend the rectangle into a cube for the z-axis
- Add some toggle to turn the machine travel on and off

FlightFixit, 2018-05-22:
This would be amazing! I need this extremely badly! It would also be helpful if when the user zeros out the machine's work coordinates the rectangle wouldn't move (it would only work in machine coordinates).

TobyGWilliams, 2018-05-23:
Thank you for the encouragement.

At the moment, I suspect there will be only one way of making it work logically (using the difference between the machine co-ordinates and the work co-ordinates). I'm open to suggestions.

It's taking me longer to learn react and how it's being used here than I thought.

TobyGWilliams, 2018-06-25:
I've created a branch with some limited preview here: https://github.com/tobygwilliams/cncjs

TobyGWilliams, 2018-06-27:
Thanks, it would be useful to know if I am providing code in the right way to make life easy for you (Am I using git correctly?). 

I agree with the settings instead of widget point, I just understand even less about how the settings section works than the widgets. I will look into the settings section.

I would also like to see if I can provide some nice code to the G-code widget to highlight where the limits of the gcode exceed the machine travel. Hence wanting to understand the widget system.

Did you see the allowance to the workspace json that I made to hold the machine configuration?

I guess there should be some allowance to turn on and off the limits rectangle here: 

[image]

cheton, 2018-06-27:
> I would also like to see if I can provide some nice code to the G-code widget to highlight where the limits of the gcode exceed the machine travel. Hence wanting to understand the widget system.

I will deprecate the G-code widget since CNCjs v2.0 to leave more space for other widgets. The visualizer can be used to display axis limits, G-code program boundary, and running status.

> Did you see the allowance to the workspace json that I made to hold the machine configuration?

Yes. I just have a question. What's the purpose of the **spindle.maxSpeed** settings?
```js
machines: [
    {
        name: 'default machine',
        travel: {
            enabled: true,
            x: { enabled: true, min: 0, max: 260 },
            y: { enabled: true, min: 0, max: 130 },
            z: { enabled: false, min: 0, max: 100 }
        },
        spindle: {
            maxSpeed: 3000
        }
    }
]
```

> I guess there should be some allowance to turn on and off the limits rectangle here:

Yes, it's doable. I will also add a dropdown for selecting a machine profile.

TobyGWilliams, 2018-06-27:
Ok, understand about gcode widget. This is still a learning exercise so I don't mind the extra work.

The spindle thing is there to 
a) be an example of using the structure to store information about the machine.
b) we have two machines, the gcode is compatible apart from the very different spindle speeds. Could be a useful check, see if you try 24,000 rpms on a 3,000 rpms machine.

I will put a table + modal thing into the settings to admin the machines.

TobyGWilliams, 2018-07-03:
I have put some commits to add a machine section to the settings, I will figure out the modal things next.

Toby

TobyGWilliams, 2019-01-14:
I have completed version 1. I have expanded the travels for limits in X, Y and Z. These are visualised as a cube, allowing for any angle or orthographic project to see where the machine and code is in the travel of the machine.

The settings area has been updated to control figuration of the machines and a UI element added to switch between the various machine configurations.

[image]

[image]

Things left to do:
- Make the machine travel compatible with any two pairs of the axis data to show only a rectangle.
- Make sure that changes to the store happen correctly on version migration.
- General debugging & testing
 

The branch is here: https://github.com/TobyGWilliams/cncjs/tree/machine-travel

MitchBradley, 2019-01-14:
Nice!

cheton, 2019-04-03:
@TobyGWilliams 

Thank you for initiating this. It's now available in 1.9.18.

closed 2019-04-03 by cheton