cncjs/cncjs #326: NPM: JSON error, cannot install

openbuilds-engineer, 2018-05-11:
#### Description

Followed https://cnc.js.org/docs/rpi-setup-guide/

#### Versions

CNCJS: None: Fails at sudo npm install -g cncjs@latest --unsafe-perm
```
pi@BIGCNC:~ $ npm -v
node 4.2.0
pi@BIGCNC:~ $ node -v
v7.10.1
```

#### How Do You Install CNCjs?

- [x] NPM sudo npm install -g cncjs@latest --unsafe-perm
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

Irrelevant

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

SCREENSHOT 
[image]

cheton, 2018-05-19:
Try updating npm installer itself to the latest version (6.x), and removing the cache (~/.npm/) before installing CNCjs again.

```sh
$ npm i -g npm
$ rm -rf ~/.npm/
```

openbuilds-engineer, 2018-05-19:
Removing cache did the trick. Thanks, we can close this one

AustinSaintAubin, 2018-05-20:
Added this problem and solutions to Wiki: https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Package-Manager-*(Recommended)*#json-error

closed 2018-05-20 by cheton