cncjs/cncjs #327: Feature request:  Decimal Imperial units on jogging button

itman496, 2018-05-14:
#### Description

Simply requesting that  I can set decimal increments instead of fractional on the jogging controls. I'm used to pressing a button to set it to jog 0.2in, for example.  Now I'm presented with crazy fractions like 1/128" and its a bit odd.  I don't believe it used to be like this in older versions, I do not remember running into this before.  Is there a way to get it back?
 
#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x (latest?)
- NPM: 5.x (latest?)

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-05-22:
Sorry, it's my fault. I'm not familiar enough with the imperial system so I don't know the expectation for particular preferences. I will add decimal options back so you can select between fractional and decimal units for jogging control.

itman496, 2018-05-22:
Not a problem!  Yeah, in imperial from my experience, no machinist ever works in fractional, only decimal.  I wish we could just do metric but.. well, you know how it is.

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)

closed 2019-03-07 by cheton