cncjs/cncjs #328: Non-obvious gotcha with CNCjs and the default gShield settings

justinclift, 2018-05-14:
#### Description

When CNCjs is used with the default g2core settings for a gShield, all CNCjs jog commands are "non-functional".  Even though the job buttons are enabled, pressing them doesn't result in any movement.

The cause turned out to be pretty simple. :smile:

The default gShield settings (in g2core) have all the axis disabled:

* https://github.com/synthetos/g2/blob/80f3fcc77f9dde29fae5d9cb6f870fa2f9bd659c/g2core/settings/settings_default.h#L403
* https://github.com/synthetos/g2/blob/80f3fcc77f9dde29fae5d9cb6f870fa2f9bd659c/g2core/settings/settings_default.h#L444
* https://github.com/synthetos/g2/blob/80f3fcc77f9dde29fae5d9cb6f870fa2f9bd659c/g2core/settings/settings_default.h#L485

Nothing in the CNCjs GUI seems to display that to the user.  This led to several hours of frustration
when trying CNCjs (a while back) before moving on to other solutions.

This time around I've investigated and found the root cause.

Manually enabling the axis via the Console works (eg `$xam=1`, `$yam=1`, `$zam=1`).  As would compiling g2core with the axis enabled in the settings.

Would it be feasible for CNCjs to check for this problem, and somehow alert the user to it?  And/or have a way to enable/disable the axis through the UI (instead of via Console)?

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.14.0  <-- as per CentOS 7
- NPM: 3.10.10  <-- as per CentOS 7

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] TinyG/g2core

#### Hardware

- [x] Arduino Due with gShield

#### Operating System

- [x] Not Applicable

cheton, 2018-05-19:
Not sure if it is feasible to check axis mode during startup, it needs to check firmware version (v0.98 or later) before sending query commands.

Adding UI options is more easy to do, it can be added to the TinyG widget for changing the axis mode: disable, standard, inhibited, radius mode.

https://github.com/synthetos/g2/wiki/Configuring-0.99-Axes#xam---axis-mode
* {xam:0} Disable. All input to that axis will be ignored and the axis will not move.
* {xam:1} Standard. Linear axes move in length units. Rotary axes move in degrees.
* {xam:2} Inhibited. Axis values are taken into account when planning moves, but the axis will not move. Use this to perform a Z kill or to do a compute-only run.
* {xam:3} Radius mode. (Rotary axes only) In radius mode gcode values are interpreted as linear units; either inches or mm depending on the prevailing G20/G21 setting. The conversion of linear units to degrees is accomplished using the radius setting for that axis. See $aRA for details.

justinclift, 2018-05-19:
Ahhh, the UI options approach might be workable then.  Not sure how best to present those four in the widget though.

The Power Management section of the TinyG widget has "Enable Motors" and "Disable Motors" buttons already:

[image]

Maybe something like that, but in a section called (say): "Axis status"?

With 1 line per axis (eg "X") and a drop thing menu showing the current status and allowing that to be changed?  Can whip up a wireframe pic pretty easily if that'd be useful. :smile:

Actually... @MitchBradley, do you have suggestions for this?  Remembering you use both CNCjs + g2 as well. :smile:

MitchBradley, 2019-06-27:
This seems like a can of worms to me.  The set of things that can be misconfigured in g2core is virtually endless, so expecting cncjs to check for all of them seems like a never-ending nightmare.  Also considering that the authors of g2core have gone silent - no updates for a long time - it just doesn't seem like a good use of time to add g2core-specific diagnostic features to cncjs.

closed 2019-06-27 by MitchBradley

justinclift, 2019-06-27:
> considering that the authors of g2core have gone silent

Yeah, I've been attempting to get in contact with them too, but radio silence. :frowning: