cncjs/cncjs #329: "Sign in" required when no users are defined

justinclift, 2018-05-17:
#### Description

Started up the CNCjs application today, after trying it out on Windows successfully a few times yesterday.

Today, it asked for a user/password login.  Which is weird, as I haven't created any users.

Looking through the docs, it points to the location of the .cncrc for Windows.

The (complete) contents of it are:

```
{                                                                                                                              
    "state": {                                                                                                                 
        "checkForUpdates": true,                                                                                               
        "controller": {                                                                                                        
            "exception": {                                                                                                     
                "ignoreErrors": false                                                                                          
            }
        }
    },
    "secret": "$2a$10$ylEMrF9a039LDvlmIdWZTe"
}
```

At a guess, something caused a "secret" to be generated and stored in the file, even though I've not defined any users.  From memory of yesterday, I did remove several of the widgets by using "Remove Widget" in their panel options.  Nothing else springs to mind that might have affected settings though.

#### Versions

- CNCjs: 1.9.15

#### How Do You Install CNCjs?

- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] TinyG/g2core

#### Hardware

- [x] Desktop or Laptop

#### Operating System

- [x] Windows 7 x64

cheton, 2018-05-19:
The secret key will be generated at the first time when creating the .cncrc file, it is mainly purposed for generating (or validating) an access token with user id and user name. If no users are configured in the `.cncrc` file, the server will return a dummy access token (i.e. empty user id/name) to bypass the **Sign In** page. See https://github.com/cncjs/cncjs/blob/master/src/app/api/api.users.js#L74-L84

Not sure what's going wrong with your `.cncrc` file. You can try removing the file to check if it works for you.

justinclift, 2018-05-19:
Yeah, that's what I did.  Killed the file, and restarted the cncjs server.  It then came up correctly, not requiring a user/pass login.

Should we kill this as it's easy to fix, or leave it open as something to be investigated? :smile:

cheton, 2018-05-20:
I will close this issue as it's fixed in your environment, and I never heard others who might face the same problem. Next time when it happened again, please keep your .cncrc file and upload the file here for further investigation. Thank you.

closed 2018-05-20 by cheton

swittman, 2025-02-08:
I've encountered this issue today when updating from 1.9.(44?) to latest (1.10.5). I have no .cncrc file, as in there wasn't one to check or delete. I do have a .cncjs/cncrc.cfg, which contains a secret, but no users field. The previously linked file no longer exists. Please advise. Thanks!