cncjs/cncjs #330: Minor note about desktop app installation in OS X

brucedesertrat, 2018-05-18:
#### Description

Running the desktop app for the first time in OS X, simpler process

#### Versions

Any unsigned OS X app can be run the first time without disabling gatekeeper (as shown in your documentation) by navigating to the ap in the Applications folder, control- or right-clicking on it and selecting 'Open' Instead of getting the 'This app is unsigned' error and not running, you will be told the App is unsigned, do you want to run it?:

[image]

[image]

This is safer than changing the Gatekeeper prefs, because it's stupidly easy to forget to set them back, and is the preferred method per Apple's Gatekeeper documentation 

https://support.apple.com/en-us/HT202491

cheton, 2018-05-19:
It looks much better than changing Gatekeeper options. Would you mind updating the wiki page with the preferred method? Thank you.
https://github.com/cncjs/cncjs/wiki/electron-app#important-notice

cheton, 2019-04-20:
Updated in the wiki page.

closed 2019-04-20 by cheton