cncjs/cncjs #331: raspberry pi model B+ offline installation help

nesquik011, 2018-05-20:
hello i am new with raspberry pi , is there away to download the files from github then manuilly install it from raspberry pi's RASPBIAN STRETCH WITH DESKTOP ? if yes someone help

AustinSaintAubin, 2018-05-21:
I recommend that you have an active internet connection durring the install & setup.
After setup is complete, you no longer have to stay online.

-------------------------------------------------------------
There is no easy way to conduct installs offline for everything.
[image]

nesquik011, 2018-05-21:
I installed online before but i want to install offline anyway ?

[image]<https://user-images.githubusercontent.com/969780/40285877-c08590b8-5c66-11e8-8540-06aacde64155.png>

cheton, 2018-05-21:
@nesquik011 
You may have to turn back to the solution of creating a pre-installed image (https://github.com/cncjs/cncjs/issues/310), and flash the image to the SD card. This is only useful when you want to do volume installation, not one-time installation.

nesquik011, 2018-05-21:
@cheton   so thats the only way ?

cheton, 2018-05-22:
Just like @AustinSaintAubin said, it's recommended that you have an active Internet connection for the initial setup.

closed 2018-05-22 by cheton