cncjs/cncjs #332: can't enter into dev mode

Thrillz, 2018-05-21:
#### Description

I installed cncjs using the npm method mentioned in the documentation. I would like to add a custom widget for automatic tool change and following the instruction on the widget boilerplate i always get this error when i run "npm run dev". I can't even find the "src" directory in "node_modules > cncjs" where exactly can i find it?

#### Versions

- CNCjs: 1.9.15
- Node.js: 8.11.2
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

PS C:\Users\vic_t\Desktop\Html folder\cncjs> npm run dev

> cncjs@1.9.15 dev C:\Users\vic_t\Desktop\Html folder\cncjs
> npm run build-dev && npm run start-dev

> cncjs@1.9.15 prebuild-dev C:\Users\vic_t\Desktop\Html folder\cncjs
> bash scripts/prebuild-dev.sh

scripts/prebuild-dev.sh: mkdir: command not found
scripts/prebuild-dev.sh: rm: command not found
scripts/prebuild-dev.sh: sh: No such file or directory
C:/Users/vic_t/Desktop/Html folder/cncjs/src C:/Users/vic_t/Desktop/Html folder/cncjs
scripts/prebuild-dev.sh: cp: command not found
scripts/prebuild-dev.sh: sh: No such file or directory
C:/Users/vic_t/Desktop/Html folder/cncjs

> cncjs@1.9.15 build-dev C:\Users\vic_t\Desktop\Html folder\cncjs
> concurrently --kill-others-on-fail --names "build-dev-app,build-dev-web" "npm run build-dev-app" "npm run build-dev-web"

[build-dev-app]
[build-dev-app] > cncjs@1.9.15 build-dev-app C:\Users\vic_t\Desktop\Html folder\cncjs
[build-dev-app] > webpack --config webpack.appconfig.development.js && npm run build-i18n-app
[build-dev-app]
[build-dev-web]
[build-dev-web] > cncjs@1.9.15 build-dev-web C:\Users\vic_t\Desktop\Html folder\cncjs
[build-dev-web] > npm run build-i18n-web
[build-dev-web]
[build-dev-web]
[build-dev-web] > cncjs@1.9.15 build-i18n-web C:\Users\vic_t\Desktop\Html folder\cncjs
[build-dev-web] > i18next-scanner --config i18next-scanner.webconfig.js "src/web/**/*.{html,js,jsx}" "!src/web/i18n/**" "!**/node_modules/**"
[build-dev-web]
[build-dev-web] [i18next-scanner] transform: count=3, file="components\\Paginations\\index.js"
[build-dev-web] [i18next-scanner] transform: count=13, file="containers\\Header\\Header.jsx"
[build-dev-web] [i18next-scanner] transform: count=12, file="containers\\Header\\QuickAccessToolbar.jsx"
[build-dev-web] [i18next-scanner] transform: count=7, file="containers\\Login\\Login.jsx"
[build-dev-web] [i18next-scanner] transform: count=2, file="containers\\Settings\\About\\AboutContainer.jsx"
[... 99 lines cut ...]
[build-dev-web] npm ERR! A complete log of this run can be found in:
[build-dev-web] npm ERR!     C:\Users\vic_t\AppData\Roaming\npm-cache\_logs\2018-05-21T09_12_36_589Z-debug.log
[build-dev-web] npm run build-dev-web exited with code 2
--> Sending SIGTERM to other processes..
[build-dev-app] npm run build-dev-app exited with code 1
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.15 build-dev: `concurrently --kill-others-on-fail --names "build-dev-app,build-dev-web" "npm run build-dev-app" "npm run build-dev-web"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.15 build-dev script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\vic_t\AppData\Roaming\npm-cache\_logs\2018-05-21T09_12_36_764Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.15 dev: `npm run build-dev && npm run start-dev`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.15 dev script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\vic_t\AppData\Roaming\npm-cache\_logs\2018-05-21T09_12_36_788Z-debug.log

cheton, 2018-05-21:
I recommend you installing "Git for Windows" (https://git-scm.com/download/win). It provides a BASH emulation (git-bash) used to run Git from the command line. This will resolve several issues on Windows environment.

Thrillz, 2018-05-21:
I have got for Windows installed. Same error message.

nesquik011, 2018-05-21:
@Thrillz   may i have the auto tool change widget please ?

nesquik011, 2018-05-21:
@Thrillz  i have been searching for you for ages

Thrillz, 2018-05-21:
I planned to test out with the widget boilerplate before making the atc
widget. So far I can't get the widget boilerplate to work. No widget yet.

nesquik011, 2018-05-21:
@Thrillz   may  i have your facebook or email ?  i want to stay updated  please , and will you make it free for popular ?

cheton, 2018-05-21:
@Thrillz 

According to the attached log, you were using Windows PowerShell command line, not Git Bash.
A Git Bash terminal should look like below:
[image]

Thrillz, 2018-05-21:
@cheton  same difference :/
[image]

cheton, 2018-05-21:
You may have to run the installation again, I guess the Unix tools was not installed properly.

When asking whether to install additional Unix utilities, choose **Use Git from the Windows Command Prompt** or **Use Git and optional Unix tools from the Windows Command Prompt**:

[image]

Choose **Checkout as-is, commit Unix-style line endings**:
[image]

After that, you can open **Git Bash**, and check whether Unix commands are fully supported:
[image]

nesquik011, 2018-05-21:
@Thrillz   plz respond

Thrillz, 2018-05-21:
@cheton once again same thing happened and i dont know why i even force cleaned cache and reinstalled everything again.

[image]
[image]

Thrillz, 2018-05-21:
@nesquik011 if i do manage to get it setup and working i will link you to the branch. This is open source i believe so no money required here, we are all trying to get stuff done because we care about it.

cheton, 2018-05-22:
@Thrillz 

Try changing the clean script in package.json
```json
"clean": "bash -c 'rm -rf ./dist ./output'",
```

and replace it with:
```json
"clean": "rimraf ./dist ./output",
```

This is a cross-platform version of `rm -rf` for Node.js.

Thrillz, 2018-05-24:
Hi @cheton i added the change you suggested and got even more errors.

[image]

[image]

Thrillz, 2018-05-24:
@cheton  if you do have an atc widget already done and functional can you please push that as a beta update so we can use that instead because i believe a number of people want this functionality but can't get past the errors in trying to get the widget in.  I need your help and i can hunt for bugs in the program as this is not for commercial use.

cheton, 2018-05-24:
@Thrillz 

These errors are probably due to the installation settings when configuring the line ending conversions:

You have to choose **Checkout as-is, commit Unix-style line endings** during installation, and run "git clone" to clone all repos again:
[image]

Thrillz, 2018-05-24:
Hi @cheton i have re-installed git as you instructed, also cleaned cache, and git cloned all the repos again, i still run into the same issue, i used a different windows machine, same set-up as mine and got the same error again. I can't run "npm run prepublish", i get this error now:

"
$ npm run prepublish
npm ERR! missing script: prepublish

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\vic_t\AppData\Roaming\npm-cache\_logs\2018-05-24T09_00_38_470Z-debug.log
"
so i don't know what else i'm doing wrong. "npm run build-dev" returns same error also.

cheton, 2018-05-24:
The "**prepublish**" script had been renamed to "**prepare**" long time ago. Instead, you have to run "npm run prepare" to make a production build:

https://github.com/cncjs/cncjs/blob/master/package.json#L28
```
"prepare": "npm run clean && npm run lint && npm run test && npm run build",
```

For running a local development server, you only need to run "npm run dev" after npm installation:
https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-local-development-server

If you still get the line ending errors reported on Windows, try modifying the ".eslintrc" file as below to suppress "linebreak-style" errors:

```json
{
    "extends": "trendmicro",
    "parser": "babel-eslint",
    "env": {
        "browser": true,
        "node": true
    },
    "settings": {
        "import/resolver": {
            "webpack": {
                "config": "../webpack.webconfig.base.js"
            }
        }
    },
    "rules": {
        "react/jsx-no-bind": [1, {
            "allowArrowFunctions": true
        }],
        "linebreak-style": 0,
        "react/prefer-stateless-function": 0,
        "react/no-access-state-in-setstate": 0
    }
}
```

Thrillz, 2018-05-24:
Hi @cheton thanks for the help, I have my interface for the auto-tool change. I did read a thread where you linked to a macro and I'd like my tool change interface to run those functionalities because at the moment it's just an interface that does nothing. Any idea how i can go about this.

nesquik011, 2018-05-24:
Hey did it work ? And show me picture of the wedges plz and can it work with normal air auto tool changer spindle

closed 2019-05-09 by cheton