cncjs/cncjs #333: Auto run

nesquik011, 2018-05-21:
how to make cncjs  auto run ?  like i turn the pi and wait few sec then boom it auto run the server i dont need to do it manually

saxa, 2018-05-21:
Depending on the linux distribution you run on your Pi. It should be just a matter of telling it start the
server up at run time. If you give us more details we can try to answer it with more details.

nesquik011, 2018-05-21:
1- i use raspberry pi 3 b+ 
2-RASPBIAN STRETCH WITH DESKTOP   
    Version:April 2018
    Release date:2018-04-18
    Kernel version:4.14
3- i am new with pi :/  linux in general i struggle installing cncjs , i did once it worked well , then i formated the SD card and re do the thing from the begging to get used to it 

so tell me what should i write or do exactly , thank you for you help  <3

AustinSaintAubin, 2018-05-21:
Have a look at the Wiki.
https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Install-Node.js-via-Node-Version-Manager-(NVM)#auto-start-on-boot-with-cron-alternative

closed 2018-06-01 by cheton