cncjs/cncjs #337: backlash compensation

nesquik011, 2018-05-30:
Hello , i am getting gearbox with nice arcmin around 5 , anyways i am using it with turn table , in the center there will be no problem and the material i am using will be a disk like its 10 cm and at the end of it , it will be 0.137 mm error coz of backlash , so is there a wedgit fix that ? So when it leaves the center it modifies its postion to be perfect ? And if nt make you add that finction

MitchBradley, 2018-05-31:
Backlash compensation is very difficult to do correctly, and really should be done in the controller, not the UI.

nesquik011, 2018-05-31:
How in the controller i am using gearbox arcmin 5 so long the axis it will be perfect 0 but when you get away for example 5 cm it will gives you 0.0137 so i think the cncjs should know that and adjust itself so when it follow the gcode at 5 cm it knows that will be error so move little up

ril3y, 2018-05-31:
In the firmware not a gear box or ui.  Meaning in the computer that is
generating the steps.  For example you should see @Alden Hart
<aldenhart08@gmail.com> 's technical break down about the complexity of
backlash compensation.

https://github.com/synthetos/g2/issues/67#issuecomment-87202803

Riley

nesquik011, 2018-05-31:
cool but i am using smoothie board , i would jump on syntheotos soon once i am done with smoothie , i want to test everything as much as i could so  someone help plz @ril3y

nesquik011, 2018-05-31:
and i am using Turn  like 4 axis

MitchBradley, 2018-05-31:
This is not cncjs's problem, so you should ask for help elsewhere.  cncjs does not know nearly enough about machine characteristics to do backlash compensation correctly.  That knowledge is at the firmware level.  Adding that level of knowledge to cncjs would make cncjs much more complicated, and more subject to timing issues.  The probability of introducing errors is very high.

cheton, 2018-06-01:
Duplicate of https://github.com/cncjs/cncjs/issues/272

cheton, 2018-06-05:
Have someone ever tried harmonic drive gear with stepper motor? This is probably a good solution to reduce or eliminate backlash.

https://www.youtube.com/watch?v=3mWemlMEzFk
http://www.harmonicdrive.net/products

nesquik011, 2018-06-06:
amazing , thank you , i will contact them

https://www.youtube.com/watch?v=3mWemlMEzFk

[https://www.bing.com/th?id=OVP.yt3pt0W6widUdk2YnVWLigHgFo&pid=Api]<https://www.youtube.com/watch?v=3mWemlMEzFk>

Harmonic Drive® strain wave gear - no backlash<https://www.youtube.com/watch?v=3mWemlMEzFk>
www.youtube.com
Harmonic Drive® strain Wave Gear Principle: http://www.harmonicdrive.net Harmonic Drive® strain wave gearing utilizes a unique operating principle which is based ...

http://www.harmonicdrive.net/products

nesquik011, 2018-06-06:
bad news :D  for everyone had hope in harmonic drive  its  smallest ratio 50:1  , which means you motor must spin 50 times ! to make one rotation , so this works with servo

closed 2018-07-06 by cheton