cncjs/cncjs #338: g2core 101 changes (TinyG Controller)

amx1, 2018-06-02:
#### Description

changed parameters of g2core version 101

spindle control, commands spe and spd not implemented, replaced by spc
coolant control, com 

Update of TinyG.js

                    // Spindle control, 0=off, 1 = CW, 2 = CCW
                    'spc': (target, val) => {
                        if (val === 0) { // Spindle Off
                            _.set(target, 'modal.spindle', 'M5');
                        } else {
                            _.set(target, 'modal.spindle', (val === 1) ? 'M3' : 'M4');
                        }
                    },
                    // Spindle speed
                    'sps': (target, val) => {
                        _.set(target, 'sps', val);
                    },

Update TinyGController.js

sr = {
.....
        mpoc: true,
        spc: true, // Spindle control (edge-101.03)
        sps: true,
        cof: true,
        com: true // Mist coolant (edge-101.03)
    };

        // Check whether the spindle and coolant commands are supported
        send('{spc:n}');
        send('{com:n}');
        send('{cof:n}');

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 6.0.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

amx1, 2018-06-02:
diff --git a/src/app/controllers/TinyG/TinyG.js b/src/app/controllers/TinyG/TinyG.js
index 2d53957..8963b86 100644
--- a/src/app/controllers/TinyG/TinyG.js
+++ b/src/app/controllers/TinyG/TinyG.js
@@ -419,28 +419,12 @@ class TinyG extends events.EventEmitter {
                         }[val] || '';
                         _.set(target, 'modal.path', gcode);
                     },
-                    // Spindle enable
-                    'spe': (target, val) => {
-                        _.set(target, 'spe', val);
-
-                        const spe = _.get(target, 'spe', 0);
-                        const spd = _.get(target, 'spd', 0);
-                        if (!spe) {
-                            _.set(target, 'modal.spindle', 'M5');
-                        } else {
-                            _.set(target, 'modal.spindle', (spd === 0) ? 'M3' : 'M4');
-                        }
-                    },
-                    // Spindle direction
-                    'spd': (target, val) => {
-                        _.set(target, 'spd', val);
-
-                        const spe = _.get(target, 'spe', 0);
-                        const spd = _.get(target, 'spd', 0);
-                        if (!spe) {
+                    // Spindle control, 0=off, 1 = CW, 2 = CCW
+                    'spc': (target, val) => {
+                        if (val === 0) { // Spindle Off
                             _.set(target, 'modal.spindle', 'M5');
                         } else {
-                            _.set(target, 'modal.spindle', (spd === 0) ? 'M3' : 'M4');
+                            _.set(target, 'modal.spindle', (val === 1) ? 'M3' : 'M4');
                         }
                     },
                     // Spindle speed
diff --git a/src/app/controllers/TinyG/TinyGController.js b/src/app/controllers/TinyG/TinyGController.js
index 167608e..6d90246 100644
--- a/src/app/controllers/TinyG/TinyGController.js
+++ b/src/app/controllers/TinyG/TinyGController.js
@@ -113,11 +113,10 @@ class TinyGController {
         mpoa: true,
         mpob: true,
         mpoc: true,
-        spe: true, // Spindle enable (edge-082.10)
-        spd: true, // Spindle direction (edge-082.10)
-        sps: true, // Spindle speed (edge-082.10)
-        cof: true, // Flood coolant (edge-082.10)
-        com: true // Mist coolant (edge-082.10)
+        spc: true, // Spindle control (edge-101.03)
+        sps: true,
+        cof: true,
+        com: true // Mist coolant (edge-101.03)
     };
     timer = {
         query: null,
@@ -673,8 +672,10 @@ class TinyGController {
         // in milliseconds (50ms minimum interval)
         send('{si:100}');
 
-        // Check whether the spindle enable command is supported
-        send('{spe:n}');
+        // Check whether the spindle and coolant commands are supported
+        send('{spc:n}');
+        send('{com:n}');
+        send('{cof:n}');
 
         // Wait for 500ms to examine supported status report fields
         await delay(500);

amx1, 2018-06-02:
[TinyG-g2core-101.zip](https://github.com/cncjs/cncjs/files/2065228/TinyG-g2core-101.zip)

cheton, 2018-06-03:
Hi @amx1 

Thank you for providing the fixes. Not all users are using the latest firmware build, so I'd prefer using feature detection rather than dropping support for `spe` and `spd`.

You can check out my last commit https://github.com/cncjs/cncjs/commit/2741fff96902fb9e74ad18fd3863e533ff05e475 in the [dev-338-g2core-fb101-spc](https://github.com/cncjs/cncjs/tree/dev-338-g2core-fb101-spc) branch.

amx1, 2018-06-03:
@cheton 

Hi cheton, good point. I checked your last commit. Did not find any errors.

Best

justinclift, 2018-06-17:
@aldenhart @giseburt This issue in CNCjs - with [proposed fix already](https://github.com/cncjs/cncjs/commit/2741fff96902fb9e74ad18fd3863e533ff05e475) - raises the same question as from a few weeks ago "How do we proactively let people know when g2core comms change?".

@cheton @amx1 We recently [created a mailing list](https://lists.links.org/mailman/listinfo/g2core) - super low volume so far - for g2core discussion.  Kind of thinking that if interested people jump on that, we could at least proactively announce protocol changes through that.  And have some discussion ahead of time in case there's a better way.  Thoughts? :smile:

cheton, 2018-06-27:
Just subscribed to the mailing list. Thank you for the notice.

justinclift, 2018-06-27:
You're welcome.  Our Roadmap page might be useful too:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/Roadmap

We try to keep that up to date as well. :smile:

cheton, 2018-06-27:
https://github.com/synthetos/g2/wiki/Roadmap#remove-queue-reports-qrn

Just have one concern about removing the Queue Reports {qr:n} command, as long time ago we tried implementing the line mode protocol, but it may still scramble the planner buffer with G2/G3 arcs. I'm not sure if the issue has been resolved with current firmware builds, I just wondered if it's safe to switch to using the line mode protocol. See https://github.com/cncjs/cncjs/issues/106#issuecomment-277436604

@amx1 What is your thought on this?

giseburt, 2018-06-27:
> it's safe to switch to using the line mode protocol. See #106 (comment)

You shouldn’t be able to scramble the planner buffer, unless you are
sending gcode wrapped in JSON, which shouldn’t be done externally.

Are you using {gc:”...”}?

justinclift, 2018-06-27:
> Just subscribed to the mailing list. Thank you for the notice.

@cheton As a heads up, you're not on the mailing list roster yet (I'm an admin for the mailing list, so can see it).  In theory it should send you a "please click this link to confirm your email" kind of thing.  Might be in your spam folder (fairly common).  If you don't receive it, let me know.  It'll mean something probably needs looking into on the server. :wink:

amx1, 2018-06-30:
Hi Rob,

 

the implementation of cncjs needs the qr information to control the flow of the gcode send routine. It will  wait and delay sending the next command if qr is below a threshold.  There is no polling of the status. It is just analysing the reporting of the controller. I can´t remember that cncjs is using {gc:”...”}.

It is a bit ago and I don´t remember the details anymore. It might be a timing issue of the line protocol that will lead to a planner buffer overflow and therefore needs this type of double check. If you send a gcode to the controller, you have to wait for an acknowledge until you send the next command. (very simple and stable handshake protocol) In case of an ARC command or other commands that generate more than one entry in the internal planner queue there might be some race conditions. If the acknowledge from the controller is send to the sender before the ARC command is completely placed in the planner queue internally, you might run into problems.

 

We have to check if the line received is sent after the ARC command is queued completely or even earlier. It is just what I remember when I helped cheton a bit to improve the cncjs sender routine.

 

Best Axel

giseburt, 2018-06-30:
I quickly scanned your code and I don’t see any use of {gc:n}, so you
shouldn’t have any buffer overflows with line mode.

However, I only see TinyG, but no mention of g2core. The protocol has
diverged, and your statements about arcs and line mode may be correct for
TinyG.

For g2core you should be able to remove most of the high-water/low-water
business and simply send a line when you get a response. There will soon be
one exception that we are still working out the details for how to handle,
and that’s the few commands that return multiple lines (such as {$:n}) ...
we are working out how to indicate that a response will continue to the
next line. We may replace the (AFAIK unused) response version with a
Boolean that says if this response is complete.

Anyway, I recommend leaving the TinyG protocol stuff as-is, and split off
the g2core protocol into its own module and then remove the qr stuff. That
should simplify that code a fair bit.

And of course if you have any questions or run into issues please let us
know.

amx1, 2018-07-04:
Dear All,

we shoud be fair. It is not my code, it is the work of cheton and I am very thankful for the work of all of you. 

Best amx1

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)

closed 2019-03-07 by cheton