cncjs/cncjs #339: what is the functions of

nesquik011, 2018-06-03:
what is the functions of ( sleep , cycle start , feedhold  ,Unlock , reset )

MitchBradley, 2018-06-03:
They are essentially artifacts of the way that GRBL works.  Feedhold is essentially like "pause" in the workflow control, but it happens immediately instead of going through all the GCode queuing.  Cycle Start is like "run" or "resume".  Unlock and Reset are more GRBL artifacts.  When you use Feedhold, GRBL goes into a safety mode where you have to explicitly unlock it before it will accept more commands.

It's all rather messy.  In cncjs-shopfloor-tablet I tried to hide all that by careful synchronization of the workflow control buttons and the GCode stream, but it isn't easy.  The fundamental challenge is the
fact that there are queuing delays between when cncjs processes a GCode line, when that line is sent to the controller, when the controller parses it, and when it is processed by the controller's motion planner, when the planned motion is transmitted to the steppers, when the motion is complete (moving takes finite time), and when cncjs is informed of the completed motion.

In an ideal world, the user interface would present a direct, zero latency view of the machine status, but that isn't feasible in the cncjs world. cncjs supports many different controllers - GRBL, tinyg, g2core, smoothie, Marlin - and next week someone will probably ask cheton to support yet another one.  Each of those controllers evolves through different software versions, with very little coordination between their developers and the UI developers.  The communication between the UI and the controller can take many forms - serial, TCP, bluetooth, etc. - with intermediaries such as Raspberry Pi's implementing the "server" component of the cncjs stack and different transports between the browser and server components.

So it boils down to this:  if you want to use cncjs, you will have to live with not-so-tight coupling between the machine and the UI.  If you need tight coupling, you are better off with an integrated machine/controller/UI setup where the same vendor is responsible for all the components.  And that will cost money.

nesquik011, 2018-06-05:
so to sum all of this up , it has no use for  smoothie or tinyg  just grbl

closed 2018-06-27 by cheton