cncjs/cncjs #340: M1 Pause Notification Blocks UI Interaction

neilferreri, 2018-06-06:
#### Description

My understanding of your handling of the M1 pause is that I should still be able to jog the machine.  When using an M1 pause in a macro (e.g. a tool change macro), a user must click "continue" to move on from the pause, so all jogging controls are unavailable.  They still appear available (the jogging controls "grey out" on an M0 pause), but they are inaccessible.  When using the keyboard to jog, all jog commands were sent after the entire macro. 

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-06-07:
Jogging buttons are disabled because Grbl is in the **Hold** state, you can enter the **Idle** state by pressing the <kbd>Cycle Start</kbd> button to enable jogging buttons.

cheton, 2018-06-07:
I think we can update the description like below:

_Click the Resume button to resume program execution. To enable jogging controls, press the [Cycle Start] button to enter the Idle state._

neilferreri, 2018-06-07:
@cheton Thank you for getting back, but I don't think I was clear.  I am talking about when an M1 is in a Macro.  The screen looks like this:
[image]
My understanding is that an M1 pause does not cause entry into a Hold state.  The jogging buttons look enabled but are blocked by the notification.  Cycle start (hardware because software is inaccessible) has no effect.

cheton, 2018-06-07:
Hmmm. All jogging and macro commands are executed through the [feeder](https://github.com/cncjs/cncjs/blob/v1.9.15/src/app/controllers/Grbl/GrblController.js#L175-L223), not the [sender](https://github.com/cncjs/cncjs/blob/v1.9.15/src/app/controllers/Grbl/GrblController.js#L253-L305), so these commands (M0/M1/M2/M30/M6) will be blocked by a modal window when executing from macros.

Instead of using a modal window to block the UI, I should use another way to control the execution, so  you can still operate the UI without blocking. I will find out a better way to make the improvement.