cncjs/cncjs #344: unable to get tinyg control of jogging buttons

retnuh22, 2018-06-24:
I have installed the desktop app on a windows 7 64 bit. (Latest version od cncjs)  install went fine.  I am able to connect to me usb port and appears to be working.  But, I have no control from laptop to controller to move my mini cnc engraver axis.  Do I need to do more to make this work.  I am not really computer programming savy.

Thanks
Retnuh22

cheton, 2018-06-27:
Have you tried turning on the **Enable hardware flow control** option when opening a serial connection from the Connection widget?

retnuh22, 2018-06-27:
Yes, the enable hardware flow control is checked. I can press the home button after connecting and the machine does Home. But I have no control over the jog feature or coordinate settings. They are grey.

cheton, 2018-06-27:
Could you take a full screenshot like below? I want to check if your machine state is ready or not.

[image]

If possible, please attach your console logs so that I will be able to know what's occurred. Just **Select All** from the dropdown, and then right click to bring up the context menu to copy text to the clipboard.

[image]

With a normal condition, you should see something like this:

```
CNCjs 1.9.15 [TinyG]
Connected to COM5 with a baud rate of 115200
{"r":{"fv":0.99,"fb":100.19,"fbs":"100.19-17-g129b","fbc":"settings_othermill.h","hp":3,"hv":0,"id":"0084-7bd6-29c6-7bd","msg":"SYSTEM READY"},"f":[1,0,1]}
feeder> {ej:1}
{"r":{"ej":1},"f":[1,0,8]}
feeder> {jv:4}
feeder> {qv:1}
{"r":{"jv":4},"f":[1,0,8]}
feeder> {sv:1}
feeder> {si:100}
{"r":{"qv":1},"f":[1,0,8]}
feeder> {spe:n}
{"r":{"sv":1},"f":[1,0,8]}
{"r":{"si":100},"f":[1,0,10]}
{"r":{"spe":0},"f":[1,0,9]}
{"sr":{"mpox":0,"mpoy":0,"mpoz":0,"ofsx":0,"ofsy":0,"ofsz":0,"g55x":0,"g55y":0,"g55z":0,"unit":1,"stat":1,"coor":2,"momo":4,"dist":0,"home":0,"mots":0,"plan":0,"line":0,"path":2,"frmo":1,"hold":0,"macs":1,"cycs":0}}
feeder> {sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spe:t,spd:t,sps:t,cof:t,com:t}}
{"r":{"sr":{"line":0,"vel":0,"feed":0,"stat":1,"cycs":0,"mots":0,"hold":0,"momo":4,"coor":2,"plan":0,"unit":1,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spe":0,"spd":0,"sps":0,"cof":0,"com":0}},"f":[1,0,219]}
feeder> {sys:n}
{"r":{"sys":{"fb":100.19,"fbs":"100.19-17-g129b","fbc":"settings_othermill.h","fv":0.99,"hp":3,"hv":0,"id":"0084-7bd6-29c6-7bd","jt":0.75,"ct":0.01,"sl":0,"lim":1,"saf":1,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":2,"spep":1,"spdp":0,"spph":1,"spdw":1.5,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":4,"qv":1,"sv":1,"si":100,"gpl":0,"gun":1,"gco":2,"gpa":2,"gdi":0}},"f":[1,0,9]}
feeder> {mt:n}
{"r":{"mt":2},"f":[1,0,8]}
feeder> {pwr:n}
{"r":{"pwr":{"1":0,"2":0,"3":0,"4":0}},"f":[1,0,9]}
feeder> {qr:n}
{"r":{"qr":48},"f":[1,0,8]}
feeder> {sr:n}
{"r":{"sr":{"line":0,"vel":0,"feed":0,"stat":1,"cycs":0,"mots":0,"hold":0,"momo":4,"coor":2,"plan":0,"unit":1,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spe":0,"spd":0,"sps":0,"cof":0,"com":0}},"f":[1,0,8]}
{"sr":{"stat":1,"momo":4,"coor":2,"plan":0,"dist":0,"frmo":1,"path":2,"posb":0,"posc":0,"mpoy":0}}
>
```

retnuh22, 2018-06-27:
I cannot figure out how to load a screen shot or pic of my screen.  Referring to your picture, my tinyg is not in a ready state as depicted by the ready next to tinyg on your screen.  I have attached the console info I get when I connect using the usb.  It also doesn't seem to have half the info yours does.  I have used the GOKO program and have been able to connect using the USB in that program.  I have also down loaded CoolTerm which allows me access to enter the tinyg and calibrate/modify.  It appears my tinyg isnt fully connectiong to the CNCjs interface.  Hopefullly its someting you can help with.

Thank you
-----------------------------------------------
CNCjs 1.9.15 [TinyG]
Connected to COM4 with a baud rate of 115200
feeder> {ej:1}
{"r":{"ej":1},"f":[1,0,7,2003]}
feeder> {jv:4}
feeder> {qv:1}
{"r":{"jv":4},"f":[1,0,7,7333]}
{"r":{"qv":1},"f":[1,0,7,6003]}
{"r":{"fv":0.970,"fb":440.20,"hp":1,"hv":8,"id":"5X0850-VCJ","msg":"SYSTEM READY"},"f":[1,0,0,6606]}
{"qr":32}
----------------------------------------------

justinclift, 2018-06-27:
Ahhh, for doing screenshots on Win 7, once you know how it's pretty simple. :smile:

To take a screenshot, press the <kbd>Print Screen</kbd> key on your keyboard.  For me, I use a desktop PC and it has one of those "full" 101-key keyboards.  On mine, the <kbd>Print Screen</kbd> key is just to the right side of my <kbd>F12 key</kbd>.

When you press that, it takes a screenshot of your whole desktop and puts it into the Windows Clipboard.

So... load up Microsoft Paint (mspaint.exe), and do the "paste" operation.  That should copy the screenshot from the clipboard into MS Paint.  Then you can just save it to a file, and drag and drop that file into a comment here.  That will add it to the comment, and show it like @cheton's one above.

Does that help? :smile:

retnuh22, 2018-06-27:
Haha. Thanks. I did the whole paint and save but when I dragged the file in it came up as a line code and nor a picture. Didn’t want to send that. Haha. Not sure if it wil convert it maybe when I hit send ?

justinclift, 2018-06-27:
Oops.  That's my bad.  I forgot to explain the end bit better. :smile:

Try dragging and dropping the picture file again.  It'll make the same line code entry looking thing as before... then click the "Preview" tab at the top of the comment box.  That should show the end result picture.  Click back on the "Write" tab to get back to the editing box where you can type stuff.

I **think** that should do the trick. :smile:

retnuh22, 2018-06-27:
Hey, Awesome !!!  Thanks 

[image]

justinclift, 2018-06-27:
And yeah, that grey-ing out of the X/Y/Z jog buttons doesn't look good.

Um, if you try giving a basic move command directly in the "Console" window tab thing, does that work?

Maybe something like this:

    G0 X50 Y50

The `G0` there just means "do a basic move", and the `X50 Y50` is just simple X + Y coordinates it's being told to move to.

If you type them into the Console area + hit enter... then in theory the router should actually move.  If it doesn't, that'll be useful info too. :wink:

justinclift, 2018-06-27:
Nah, you have all the right bits.  When I downloaded CNCjs for Win 7 a few weeks back, I used the desktop version of it as well.  For me (once I'd figured things out), that was all that's needed.

I'm not using a TinyG though, so am not exactly sure where to point you at next. :wink:

It's good the spindle moves when instructed with direct G-Code.  It seems to mean that the basic communications are working ok, there's just something that CNCjs isn't recognising, so it's not enabling the jog buttons.

I'm sure @cheton will have useful suggestions, though it might be a tomorrow thing. :smile:

cheton, 2018-06-28:
@retnuh22 

I guess your board needs to wait more than 1000ms for the bootloader to complete before sending subsequent commands. The system startup message didn't appear earlier before sending initialization commands:

```
{"r":{"fv":0.970,"fb":440.20,"hp":1,"hv":8,"id":"5X0850-VCJ","msg":"SYSTEM READY"},"f":[1,0,0,6606]}
```

You can see below to get to know what commands will be sent during initialization:

https://github.com/cncjs/cncjs/blob/ccf1d65aea3e126229ff310d21e047fc56b8d2bf/src/app/controllers/TinyG/TinyGController.js#L653-L707

After examining your attached logs, I didn't see the return result of enabling status report verbosity `{sv:1}` and status report interval `{si:100}`. Could you try manually entering below commands on the Console widget to check whether it will enable status report:

```
{sv:1}
{si:100}
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spe:t,spd:t,sps:t,cof:t,com:t}}
{sr:n}
```

cheton, 2018-06-28:
@retnuh22 

As you're using the desktop app, I need your help to provide verbose output of the CNCjs server. 

See [full instructions](https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows) below:

1. Go to https://github.com/electron/electron/releases/tag/v1.8.7, and download a zip file that matches your target platform. For example, you can download `electron-{version}-win32-x64.zip` when running the x64 version of Windows.

2. Extract the ZIP file to a folder. For example: <b>C:\\Temp\\electron-{version}-win32-x64\\</b>.
3. Go to the installation directory of cncjs: `cd C:\Users\{Username}\AppData\Local\Programs\cncjs\resources\app\`
4. Run <b>C:\\Temp\\electron-{version}-win32-x64\\electron.exe main.js -vvv</b>, then you will be able to see verbose output with `-vvv`.

    [image]

cheton, 2018-06-28:
@justinclift There is another issue #312 which is similar to this case. It's not easy to do a recovery with such condition, adding a configurable value to the connection widget that can adjust the delay might be a better way to go.

justinclift, 2018-06-28:
Ahhh yeah, that does look a similar thing.  Making that value a variable then seems like it would help out everyone hitting this problem. :smile:

cheton, 2018-06-28:
@retnuh22 

I just make a test with my Mac, connecting to an Arduino Due running with G2 firmware build 100.19. It seems that it doesn't matter with the delay, even if I set it to zero.

[image]

You can try other options to check if works:
1. Close connection and then open again
2. Turn off "Enable hardware flow control" 
3. Switch to another environment other than Windows, or use the NPM installation method
4. Upgrade firmware

MitchBradley, 2018-06-28:
My setup has a Rasp Pi Zero talking to an Arduino Due running G2.  The Arduino sometimes (often) starts in the wrong USB mode and will not communicate properly over the USB serial connection.  To solve this problem I have connected a wire from GPIO4 on the Pi to the reset pin on the Due, and added the following lines to /etc/rc.local:

```
# Drive the TinyG controller reset line low then high.  This ensures that
# it comes up in the right USB mode
echo 4 >/sys/class/gpio/export
echo out >/sys/class/gpio/gpio4/direction
echo 0 >/sys/class/gpio/gpio4/value
sleep 1
echo 1 >/sys/class/gpio/gpio4/value
sleep 4
```

MitchBradley, 2018-06-28:
Update on the previous post regarding Due reset:

That reset recipe sometimes failed to solve the problem.  I tracked it down to an ordering problem with various startup scripts.  I was starting cncjs via a crontab, but doing the reset in rc.local.  It turns out that the user crontab runs *before* rc.local.  I solved that by creating a script that does the reset dance, then starts cncjs.  The crontab @reboot entry then runs that script.  It seems to be working, but many more reboots will be necessary before I have complete confidence.

retnuh22, 2018-06-28:
Wow, great response folks.  I really appreciate you all taking the time and effort to help out..... Though all of this sounds great and I understand most of it, implementation of asked requests or diagnosis might be difficult for me with out "Lame-man" terms or steps.  Haha.  I'll re-read the previous posts and send what I can figure out.  Thanks

justinclift, 2018-06-28:
No worries at all @retnuh22.  Do your best, and don't be afraid to ask questions if/when you get stuck.

We're all volunteers helping out here, so our responsiveness can vary... but we know the frustration ourselves when things that **should** work, just plain refuse to. :wink:

retnuh22, 2018-06-28:
Ran the following in the console as asked:

{sv:1}
{si:100}
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spe:t,spd:t,sps:t,cof:t,com:t}}
{sr:n}

Got the below as a result....

{"r":{"sv":1},"f":[1,0,238,9467]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
>

retnuh22, 2018-06-28:
figured I could give the tinyg settings out of the cncjs console as a reference if it may help.  Also I ran the electron.exe after downloading but I don't know how to get to the screen shot of the "Developer Command Prompt" or how to do or get the (-vvv) ???  Sorry.
----------------------------------------------------------------
client> $$
[fb]  firmware build            440.20
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    5X0850-VCJ
[ja]  junction acceleration  100000 mm
[ct]  chordal tolerance           0.0000 mm
[sl]  soft limit enable           0
[st]  switch type                 1 [0=NO,1=NC]
[mt]  motor idle timeout          2.00 Sec
[ej]  enable json mode            0 [0=text,1=JSON]
[jv]  json verbosity              4 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[js]  json serialize style        1 [0=relaxed,1=strict]
[tv]  text verbosity              1 [0=silent,1=verbose]
[qv]  queue report verbosity      1 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[ec]  expand LF to CRLF on TX     1 [0=off,1=on]
[ee]  enable echo                 0 [0=off,1=on]
[ex]  enable flow control         2 [0=off,1=XON/XOFF, 2=RTS/CTS]
[baud] USB baud rate              5 [1=9600,2=19200,3=38400,4=57600,5=115200,6=230400]
[net] network mode                0 [0=master]
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution    4.0000 mm
[1mi] m1 microsteps               8 [1,2,4,8]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[2ma] m2 map to axis              1 [0=X,1=Y,2=Z...]
[2sa] m2 step angle               1.800 deg
[2tr] m2 travel per revolution    4.0000 mm
[2mi] m2 microsteps               8 [1,2,4,8]
[2po] m2 polarity                 0 [0=normal,1=reverse]
[2pm] m2 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[3ma] m3 map to axis              2 [0=X,1=Y,2=Z...]
[3sa] m3 step angle               1.800 deg
[3tr] m3 travel per revolution    4.0000 mm
[3mi] m3 microsteps               8 [1,2,4,8]
[3po] m3 polarity                 0 [0=normal,1=reverse]
[3pm] m3 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[4ma] m4 map to axis              3 [0=X,1=Y,2=Z...]
[4sa] m4 step angle               1.800 deg
[4tr] m4 travel per revolution  180.0000 mm
[4mi] m4 microsteps               8 [1,2,4,8]
[4po] m4 polarity                 0 [0=normal,1=reverse]
[4pm] m4 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum        900 mm/min
[xfr] x feedrate maximum        800 mm/min
[xtn] x travel minimum            0.000 mm
[xtm] x travel maximum          150.000 mm
[xjm] x jerk maximum             20 mm/min^3 * 1 million
[xjh] x jerk homing              40 mm/min^3 * 1 million
[xjd] x junction deviation        0.0500 mm (larger is faster)
[xsn] x switch min                1 [0=off,1=homing,2=limit,3=limit+homing]
[xsx] x switch max                0 [0=off,1=homing,2=limit,3=limit+homing]
[xsv] x search velocity         500 mm/min
[xlv] x latch velocity          100 mm/min
[xlb] x latch backoff             5.000 mm
[xzb] x zero backoff              1.000 mm
[yam] y axis mode                 1 [standard]
[yvm] y velocity maximum        900 mm/min
[... 98 lines cut ...]
[g59y] g59 y offset               0.000 mm
[g59z] g59 z offset               0.000 mm
[g59a] g59 a offset               0.000 deg
[g59b] g59 b offset               0.000 deg
[g59c] g59 c offset               0.000 deg
[g92x] g92 x offset               0.000 mm
[g92y] g92 y offset               0.000 mm
[g92z] g92 z offset               0.000 mm
[g92a] g92 a offset               0.000 deg
[g92b] g92 b offset               0.000 deg
[g92c] g92 c offset               0.000 deg
[g28x] g28 x position             0.000 mm
[g28y] g28 y position             0.000 mm
[g28z] g28 z position             0.000 mm
[g28a] g28 a position             0.000 deg
[g28b] g28 b position             0.000 deg
[g28c] g28 c position             0.000 deg
[g30x] g30 x position             0.000 mm
[g30y] g30 y position             0.000 mm
[g30z] g30 z position             0.000 mm
[g30a] g30 a position             0.000 deg
[g30b] g30 b position             0.000 deg
[g30c] g30 c position             0.000 deg
tinyg [mm] ok>
>

retnuh22, 2018-06-28:
Hey... Update.... I was in the tinyg console report section and saw a "Reset default" option.  I hit the option and now have full access to jogging!!  Ill test and let you guys know if its working or not.

[image]

justinclift, 2018-06-29:
@retnuh22 Cool, sounds like you're well on the way to a working solution. :smile:

MitchBradley, 2018-06-29:
My best guess for why it was not working before was that it was failing to recognize "JSON mode" commands.   The old-style way of talking to tinyg is called "text mode", with commands formed like "$ej=1". The new way is called "JSON mode", where you would instead write "{ej:1}".  JSON mode is better for the long term because it allows more complex information to be transmitted without ambiguity, and CNCjs uses it.  In theory, tinyg is suppose to enter JSON mode as soon as it sees a "{" character, but the logs you show suggest that might not be working.  That theory would explain why your GCode commands worked, because GCode can be transmitted the same way regardless of whether JSON mode is in effect.

The default settings will probably result in slow machine movement, and might result in some axes being backwards, or other mismatches with your machine.

To fix that you would need to adjust the axis rate parameters - for example "xvm" - and maybe the motor parameters - for example "1tr".  The best settings will depend on your machine, in particular how far it moves for each motor step and how fast it can go without losing precision.

MitchBradley, 2018-06-29:
1sa (and 2sa and 3sa) probably should be 1.8 degrees, because most steppers are 200 steps per revolution.

1mi depends on whether your motor drivers do microstepping.

1tr depends on the mechanical coupling between the motor and the machine - how far does the machine move for one full rotation of the motor.

Once all those parameters are set correctly, the machine should move accurately at slow speeds.

The next step is to tune the speeds so you can go as fast as possible without losing accuracy.  If you try to go too fast, or accelerate too rapidly, friction, inertia, and tool load can make the motors stall and lose position.  https://github.com/synthetos/TinyG/wiki/TinyG-Tuning has some tips.  You might also find suggestions for your particular machine on the web, from others who have used it.

MitchBradley, 2018-06-29:
I created a wiki page describing what I did to ensure proper reset of the Arduino Due connected to my CNCjs Raspberry Pi - https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Pi-with-Arduino-Due .

justinclift, 2018-06-29:
Good thinking @MitchBradley. :smile:

jimfong1, 2018-07-06:
Running into this same problem of no right side jog control with latest firmware release of tinyg2.   Hardware flow is enabled. Tried Restore Defaults. I see the json data in the console screen coming through. 

I have no problem with a older tinyg2 release and CNCjs works just fine with it.  

Windows 10, latest version of CNCjs,  tinyg2 release 101.03 (May 30th release)

I even compiled the latest Tinyg2 edge source and same problem.  I can connect to Due board with realterm and do a simple commands but CNCjs jog doesn't work. 

regards,

Jim

cheton, 2018-07-06:
@jimfong1 Please follow the same [steps](https://github.com/cncjs/cncjs/issues/344#issuecomment-400737539) to provide your screenshot and console logs, then we can check what's going wrong with your setup.

justinclift, 2018-07-06:
@jimfong1 Out of curiosity, do you remember what the release number was for the working older tinyg2 release?  That might be useful for future people when we're doing troubleshooting. :smile:

jimfong1, 2018-07-06:
The older firmware that works was downloaded from this page,  Due+gShield | edge | 078.03 
 
http://synthetos.github.io/g2/

I took apart the Due Test setup yesterday. So no screen shots until I can get back to it.  I have to work on a BeagleBone Black Beagleg test install now.  Hopefully this gets sorted out before then. 

Jim

justinclift, 2018-07-06:
Thanks @jimfong1, that's useful info. :smile:

When you get back to testing this, a screenshot will be really helpful.  It'll let us more easily know where things have gotten mucked up, and be a good point for further debugging. :smile:

MitchBradley, 2018-07-06:
Try hitting the reset button on the gShield after cncjs has loaded.

cheton, 2018-07-13:
Close this issue. You can continue to discussion here and share your findings.

closed 2018-07-13 by cheton

denverh2000, 2018-11-25:
I've been fighting exactly the same problem with a tinyg as the OP, using both a r-pi and a desktop debian system.  I tried both the regular web based cncjs and the desktop app.  I was beginning to think I was going to have to abandon the tinyg when I discovered this thread.  The restore defaults from cncjs took care of the problem for me as well.  Which is a bit odd, because I had tried that already with no success.  The only difference is that when I did the reset before, I used the manual command through a serial terminal, then connected the tinyg to cncjs.  This discussion has been very helpful.  A big thank you to everyone.