cncjs/cncjs #345: All Ok?

kikolobo, 2018-06-26:
It has been a while since a new update.

I hope all is OK 😊

Hope development doesn't halt.

cheton, 2018-06-26:
Sorry for the delay, it's been a long time since last update. I was busy with day work in the past 3~4 months, and I also need to take care of two kids under 2 every night after work, including holidays, so I didn't have much free time on my side projects.

In the following months, I will be able to relieve the heavy load conditions from day work, then I can start to resolve these pending issues. 😄

AustinSaintAubin, 2018-06-26:
@cheton I know how that home and work life goes. Glad to see you are well and taking time for family.
Please setup a patreon at some point, I would like to donate on a regular basis. 
Thanks for all that you do and have done for the CNC community with CNCjs.

kikolobo, 2018-06-26:
@cheton thank you so much, I second the donation idea. I would donate on a regular basis too. This is an amazing project that has changed my workflow tremendously for the good!

TobyGWilliams, 2018-06-27:
I hope that there will be new feature and release soon

closed 2018-07-13 by cheton

kikolobo, 2018-08-21:
No News Good News?

@cheton Please don't kill the project.. Let us know how to help to make this something of interest for you commercially. You need to make money on this so we can expect features!

Dont kill it. It is too good to die!

saxa, 2018-08-21:
Good side of open source is you can always fork and maintain the code by yourself or community. Good side of open source software is also that community can contribute with whatever they know, larger the community better the project goes.

nesquik011, 2018-08-21:
@saxa   Family

kikolobo, 2018-08-21:
That is great to hear..

I would love to help. I am not as good though :) .. What is the best path to setup a development envirnonment for CNCJS in your experience?

saxa, 2018-08-21:
I would start with installing it from source. Follow the setup on the cncjs page and you should be good. 
It is all here https://github.com/cncjs/cncjs/wiki

kikolobo, 2019-01-21:
Hi @cheton.

I hope all is well with you.. Hope to get a release soon to signal the project is still going on. 😊