cncjs/cncjs #346: Webcam stream settings return to default after reboot

thelostcrafts, 2018-06-28:
#### Description

The webcam feature works fine until a reboot of the raspberryPi occurs. At that point the settings for the web cam stream returns to default settings. The picture below is what I have set and it works perfectly, after reboot it defaults back to "Use a built-in camera or a connected webcam"

Thanks for all your amazing work on this project. 
[image]

#### Versions

- CNCjs: 1.9.15
- Node.js: 10.5.0
- NPM: 6.1.0

#### How Do You Install CNCjs?

- [] NPM
- [ ] Download the CNCjs Desktop Application
Running the latest CNCjs raspberryPi Image

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2018-06-28:
The workspace settings is stored in browser's Local Storage. If you change another browser, open window in incognito mode, or accidentally clear local storage, either one will lose previously saved workspace settings.

You can go to **Settings > Workspace** to view current workspace settings:

[image]

I don't know what may go wrong with your environment, but you can always export workspace settings after making several changes to prevent accidental data loss.

thelostcrafts, 2018-06-28:
Ahhh perfect.. thank you!