cncjs/cncjs #347: Electron App displays "Not found :(" on startup. Failing to create ./sessions directory.

mrdunk, 2018-07-04:
#### Description

App fails to start correctly with the following browser output:
```
Not found :(
Sorry, but the page you were trying to view does not exist.

It looks like this was the result of either:

a mistyped address
an out-of-date link
```
and shell output:
```
duncan@lapdancer:~$ cncjs-app 
2018-07-04T22:59:41.621Z - info: init Loading configuration from "/home/duncan/.cncrc"
Error: EACCES: permission denied, mkdir './sessions'
    at Object.fs.mkdirSync (fs.js:891:18)
    at appMain (/opt/CNCjs/resources/app/app/index.js:10595:22)
    at server (/opt/CNCjs/resources/app/app/index.js:7316:38)
    at /opt/CNCjs/resources/app/node_modules/webappengine/dist/app/app.multihost.js:132:25
    at Array.forEach (<anonymous>)
    at Object.appMain [as default] (/opt/CNCjs/resources/app/node_modules/webappengine/dist/app/app.multihost.js:97:31)
    at createServer (/opt/CNCjs/resources/app/node_modules/webappengine/dist/app/server.js:62:49)
    at serverMain (/opt/CNCjs/resources/app/node_modules/webappengine/dist/app/server.js:114:9)
    at createServer (/opt/CNCjs/resources/app/node_modules/webappengine/dist/app/index.js:57:33)
    at module.exports (/opt/CNCjs/resources/app/node_modules/webappengine/index.js:13:12)
    at Object.createServer (/opt/CNCjs/resources/app/app/index.js:7320:32)
    at /opt/CNCjs/resources/app/cnc.js:100:26
    at Promise (<anonymous>)
    at F (/opt/CNCjs/resources/app/node_modules/core-js/library/modules/_export.js:35:28)
    at cnc (/opt/CNCjs/resources/app/cnc.js:96:12)
    at _callee$ (/opt/CNCjs/resources/app/main.js:94:50)
2018-07-04T22:59:41.735Z - error: The multi-host server does not exist: {
    "type": "server",
    "route": "/"
}     at /opt/CNCjs/resources/app/node_modules/webappengine/dist/app/app.multihost.js:138:27 id=0, pid=7843
2018-07-04T22:59:41.745Z - info: init Starting the server at http://127.0.0.1:40260
```

As a (insecure) workaround,
strace indicates cncjs is running in the `/opt/CNCjs/resources/app/app` directory.
If i make that directory writable to the cncjs process it cncjs works as expected.

I suggest creating `sessions` under `~/.cncjs` 
in `/opt/CNCjs/resources/app/app/index.js:10592`.

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.14.3
- NPM: 3.10.6

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [ x] Linux

cheton, 2018-07-05:
Related to https://github.com/cncjs/cncjs/issues/319 and https://github.com/cncjs/cncjs/issues/320

cheton, 2018-07-05:
It make sense to create sessions under user's home directory. I will make the change in 1.9.16.

cheton, 2018-07-05:
Fixed in PR #348 

You can try the prebuilt image and let me know the result:
https://github.com/cncjs/cncjs/releases/tag/319-320-347-sessions-latest

mrdunk, 2018-07-05:
I'm at work just now so testing on a different machine but the results are the same;

The original release still fails on this machine.
The new release starts correctly.

I'll confirm this is the case on my machine at home this evening but i think it's safe to say this is fixed.
Thanks for the quick fix!
dunk.

closed 2018-07-05 by mrdunk

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)