cncjs/cncjs #356: Macro tool dia calc

JamieAKing, 2018-07-08:
#### Description
Using a touch plate to accurately measure tool diameter

I have an aluminium z touch off plate mounted in the bed of my machine. G30 is centered over it and I have adjusted the tool change macro accordingly. During doing this I had a thought. Would it be possible to use a macro to output a tool diameter that can then be used in the cam program. Measuring the true diameter of single flute cutters is incredible frustrating. I currently have to  cut a slot in some scrap wood and the measure that slot with calipers which isn’t easy in a 2mm slot.

So my question is: is there a way to display the stored value in a macro variable in the UI?

A quick and dirty macro would look something like this:

G53 Z-2 ; raise to tool change height 
G30 ; probe location 
M0 ; pause to change tool
G91
G38.2 Z-60 F400 ; protected move closer to touch plate
G00 Z5 ; debounce in case of contact 
G38.3 Z-50 F200 ; fast seek of touch plate 
G00 Z2 ; debounce
G38.3 Z-5 F50 ; fine seek Z0
G4 P1 ; dwell to allow position update before setting zero
G10 L20 P0 Z0 ; zero Z in current wcs 

%CENTER=posx
%TPW=40 ; touch plate width
%X_MOVE = [(TPW/2)+5]

G91 ;  still in incremental mode but an extra just to make sure 
G00 Z5
X[(X_MOVE * -1)] ; move half touch plate distance in negative 
Z-10
M3 S250 ; start spindle running slowly 
G38.3 X5 F10 ; probe very slowly as not to cut into probe plate 
M5 ; stop spindle to prevent over heat at low rpm 
G4 P0.5 ; dwell for to allow position update

%XLEFT=posx

G00 X-5 
Z10

; return to center of touch plate as distance 
; moved during probe cycle unknown 
G90 ; absolute mode
X[CENTER]

G91 ; increment mode 
X[X_MOVE]; move half touch plate distance plus 5mm 
Z-10
M3 S250
G38.3 X-5 F10
M5
G4 P0.5
%XRIGHT=posx
G00 X5
M5 
Z10
G90
X[CENTER]
G53 Z-2

%TOOLDIA = [XRIGHT - PLW - [XLEFT]

In this example a tool can be loaded and then probed automatically all I need is to see what %TOOLDIA has stored 

Any input would be greatly appreciated 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

cheton, 2018-07-10:
I will add it in a future release. For a temporary solution, you can append `G4 P0 (TOOLDIA=[TOOLDIA])` to the last line so you can see the stored value on the console widget.

JamieAKing, 2018-07-10:
This is all I need no further action required thank you. it had occurred to me to use g10 to set one of the axis to the stored value but I didn't want to do that in case I wanted to replace a broken tool mid cycle as I would loose my work offsets. I know g92 can set temporary values but I haven't used it perhaps I will experiment.

JamieAKing, 2018-07-10:
G4 P0 (TOOLDIA=[TOOLDIA]) through back error 2 bad number format
so does
G10 L20 P0 X[TOOLDIA]
Im assuming it may be too may decimal places but without seeing the stored value it is hard to know

nesquik011, 2018-07-13:
@cheton   and if you please can tell me how to add auto tool change widget ? someone did try to make it , so if you too can improve it and add it to your machine cncjs plz

cheton, 2019-04-14:
You can find a few examples at https://github.com/cncjs/cncjs/wiki/User-Guide#examples to get started. For a complete list of macros, go to https://github.com/cncjs/CNCjs-Macros.

closed 2019-04-14 by cheton