cncjs/cncjs #358: Tool not moving

nkp1215, 2018-07-11:
#### Description

ALARM:9 (Homing fail).
 
My first time starting up the CNC in some months. Before the few month hiatus, the CNC was working fine. Now I can no longer get the CNC to move. I'm using a raspberry pi and SSHing. It connects to the machine "/dev/ttyUSB0" I can unlock the machine, it will not home. If I manually type x0 y0 x0 says ok nothing moves. If I try jogging the graph window will show the dot moving but my machine doesn't move, the axes values will even change nothing moves. 

All this good stuff

$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=3 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=25.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$30=1000 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.000 (X-axis travel resolution, step/mm)
$101=250.000 (Y-axis travel resolution, step/mm)
$102=250.000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)

HELPPPPPPPPPPPPPPPPPPP!!!!!!!!!!!! PLEASE!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

#### Versions

- CNCjs: 1.9.10
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [X ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

MitchBradley, 2018-07-11:
That sounds like an electrical problem with the drivers.  Is the power supply to the driver board turned on and connected properly?

nkp1215, 2018-07-12:
Yes, I will run all electrical lines, but nothing has been disconnected.

AustinSaintAubin, 2018-07-13:
Your Hard limits are disabled... that could be a problem.
set to `$21=1` (Hard limits enable, boolean)

Do the motors lock (have power to them)?
If not then this seems like a hardware issue.

Take a look at my settings and notes of [my X-Carve & X-Controller](https://gist.github.com/AustinSaintAubin/5fb4343cc34f94578d62b1432178632b).

cheton, 2018-07-29:
Close this issue since it's probably a hardware issue rather than a software issue. You can also post your question at https://github.com/gnea/grbl/issues to seek help.

closed 2018-07-29 by cheton