cncjs/cncjs #362: ethernet

nesquik011, 2018-07-18:
does cncjs ethernet works with smoothie ? i did try to connect with ethernet it didnt work , maybe i did something wrong with the wires

cheton, 2018-07-18:
Connected to Smoothie over telnet is not supported by CNCjs 1.9, only serial connection is allowed. Adding support for Ethernet connection is already planned in CNCjs 2.0 milestone.

closed 2018-07-18 by cheton

nesquik011, 2018-07-19:
Thats amazing milestone i love that name too nice name , would love that you add them change for us too please , and there is someone already made auto tool change widget please add it tool because i cant manage to add that widget and try to work on it , thank you so much for you hard work and i would love to donate BTC got a wallet ?

nesquik011, 2018-07-19:
@cheton

piflixe, 2020-09-03:
I would also like to upvote this feature. I am having connection issues using the USB serial connection with a 6kW spindle and I think using ethernet would solve that problem