cncjs/cncjs #368: Bad Console input related to browser/platform?

fbrier, 2018-07-29:
#### Description

On Linux Mint, using the Firefox browser, connected to cncjs running on a Raspberry Pi, entering gcode commands into the Console works. The command executes. But using the Chrome browser on the same laptop does not. Attempting to use Android with the Chrome browser, or Firefox, or Opera, again does not work with the same behavior. I even tried the Silk browser and Firefox browser on a Kindle Fire.

The incorrect behavior is the cursor stays at the beginning of the prompt line, ">", and as characters are entered, they appear in sequence, but the cursor stays at the beginning. When you hit enter, it clears the letters, ie: "M115", and displays another empty prompt line, ">", and no command is executed, nor output generated.

Has anyone else seen this behavior? Do tablet browsers work fine with cncjs? Since it works fine with Firefox under Mint, it seemed like it is a browser compatibility issue. But with so many browsers doing the same thing, perhaps it is an issue with the cncjs on the Raspberry Pi.

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

I used an image from a [github project](https://github.com/jeffeb3/v1pi/), however, I previously used NPM and believe I had the same issue.

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Marlin
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Host Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

#### Browser Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux (Mint)
- [x] Android
- [x] Kindle Fire

fbrier, 2018-07-30:
I cloned the cncjs repo and found that [xterm](https://github.com/xtermjs/xterm.js) was being used as the Console widget. The project does [not currently support mobile browsers](https://github.com/xtermjs/xterm.js/issues/675). There are lots of different issues which seem to affect xterm's behavior including [lang](https://github.com/xtermjs/xterm.js/issues/547), [character set](https://github.com/xtermjs/xterm.js/issues/487), which [property used to retrieve the key value](https://github.com/xtermjs/xterm.js/issues/6), [terminal lines getting rewritten](https://github.com/xtermjs/xterm.js/issues/835).

I borrowed my wife's laptop and just tested Chrome on Windows and it works fine with cncjs as well.

cheton, 2018-08-01:
There is a breaking change introduced in xterm v3.1 and it is not compatible with current implementation. I may consider rewriting the terminal or trying to find a better solution to resolve browser compatibility issues especially on tablet.

Tyriar, 2018-08-08:
@cheton FYI we're open to officially supporting mobile in xterm.js, it just needs someone to help drive it (https://github.com/xtermjs/xterm.js/issues/1101). In fact the efforts being made in https://github.com/xtermjs/xterm.js/issues/1507 will help enable plugging in different front ends more easily (if mobile changes the frontend radically).

Is this breaking change related to mobile support specifically? I'm curious what it was if you know?