cncjs/cncjs #369: windows server hosting

nesquik011, 2018-07-31:
my friend working on cncjs edit some stuff like colors etc :/ he is not prof but he is trying 

anyway we want to run that copy locally on windows , to see our results , is there away ?  if yes how ? 

i followed that videos https://www.youtube.com/watch?v=ZGu3gR9As9k&t=120s

it didnt work this error shows up 

C:\Users\dream\cncjs>node server
module.js:549
    throw err;
    ^

Error: Cannot find module 'C:\Users\dream\cncjs\server'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3

the same if i wrote node start cnc.js or run cncjs or start cncjs  ,  so help me please

nesquik011, 2018-07-31:
i did try docker too  i for link from your git hub  , docker was trying to build it gave that error Build failed: manifest for nodesource/vivid:6 not found

cheton, 2018-08-01:
I updated the [Dockerfile](https://github.com/cncjs/cncjs/blob/master/Dockerfile) several days ago. I recommend you run `docker pull cncjs/cncjs` to pull latest image from Docker Hub, or build your own image with below settings:

```
FROM node:8
MAINTAINER Cheton Wu <cheton@gmail.com>

# cache package.json and node_modules to speed up builds
ADD package.json package.json
RUN npm i npm@latest -g
RUN npm install --production
RUN npm install -g nodemon

ADD . .
EXPOSE 8000
CMD ["nodemon", "bin/cnc"]
```

cheton, 2018-08-01:
You can follow instructions in the contribution guide to run development server, make production build, or create a desktop app.

https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-local-development-server
https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-production-build
https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#build-desktop-apps

nesquik011, 2018-08-01:
@cheton    i pulled my edit copy from GitHub to docker  and changed the file  Dockerfile with 

FROM node:8
MAINTAINER "nesquik011" <dream_land_2@homail.com>

# cache package.json and node_modules to speed up builds
ADD package.json package.json
RUN npm i npm@latest -g
RUN npm install --production
RUN npm install -g nodemon

ADD . .
EXPOSE 8000
CMD ["nodemon", "bin/cnc"]

Then at Build Settings in  Type      Name      Dockerfile Location and Docker Tag Name 
                                         Branch   Master               /Dockerfile                        latest

and the build was successful this time on docker 

`Status: Downloaded newer image for nesquik011/testing1:latest
068a8e4393d0: Waiting
dream@DESKTOP-F353N57 MINGW64 /c/Program Files/Docker Toolbox
$ docker run --privileged -p 8000:8000 --rm --name cnc nesquik011/testing1:latest
[nodemon] 1.18.3
[nodemon] to restart at any time, enter `rs`
[nodemon] watching: *.*
[nodemon] starting `node bin/cnc`
module.js:549
    throw err;
    ^

Error: Cannot find module '../dist/cnc/cnc'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/bin/cnc:10:11)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
[nodemon] app crashed - waiting for file changes before starting...
[nodemon] Internal watch failed: watch /media/font-awesome/white/svg/folder.svg ENOSPC`

### ** but this happened , how to fix **

cheton, 2018-08-01:
You have to run `npm run prepare` prior to the Docker build, or you will get empty dist folder.

nesquik011, 2018-08-01:
@cheton    i wrote in docker terminal   run prepare

`dream@DESKTOP-F353N57 MINGW64 /c/Program Files/Docker Toolbox
$ npm run prepare
npm ERR! path C:\Program Files\Docker Toolbox\package.json
npm ERR! code ENOENT
npm ERR! errno -4058
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open 'C:\Program Files\Docker Toolbox\package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\dream\AppData\Roaming\npm-cache\_logs\2018-08-01T11_57_55_219Z-debug.log`

nesquik011, 2018-08-01:
may you give more detail how to write it and where

nesquik011, 2018-08-01:
### **never mind i did copy the package.json that one you already made to have the same environment to docker toolbox folder then it gave me this error  **

`$ npm run prepare

> cncjs@1.9.16-pre prepare C:\Program Files\Docker Toolbox
> npm run clean && npm run lint && npm run test && npm run build

> cncjs@1.9.16-pre clean C:\Program Files\Docker Toolbox
> bash -c 'rm -rf ./dist ./output'

> cncjs@1.9.16-pre lint C:\Program Files\Docker Toolbox
> npm run eslint && npm run stylint

> cncjs@1.9.16-pre eslint C:\Program Files\Docker Toolbox
> eslint --ext .js --ext .jsx *.js src test

'eslint' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.16-pre eslint: `eslint --ext .js --ext .jsx *.js src test`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.16-pre eslint script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\dream\AppData\Roaming\npm-cache\_logs\2018-08-01T12_49_42_506Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.16-pre lint: `npm run eslint && npm run stylint`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.16-pre lint script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\dream\AppData\Roaming\npm-cache\_logs\2018-08-01T12_49_42_546Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.16-pre prepare: `npm run clean && npm run lint && npm run test && npm run build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.16-pre prepare script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\dream\AppData\Roaming\npm-cache\_logs\2018-08-01T12_49_42_590Z-debug.log

dream@DESKTOP-F353N57 MINGW64 /c/Program Files/Docker Toolbox
$ install npm
install: missing destination file operand after 'npm'
Try 'install --help' for more information.

dream@DESKTOP-F353N57 MINGW64 /c/Program Files/Docker Toolbox
$ npm install
npm WARN deprecated github@8.2.1: 'github' has been renamed to '@octokit/rest' (https://git.io/vNB11)
npm WARN deprecated babel-preset-es2015@6.24.1: 🙌  Thanks for using Babel: we recommend using babel-preset-env now: please read babeljs.io/env to update!
npm ERR! code Z_BUF_ERROR
npm ERR! errno -5
npm ERR! zlib: unexpected end of file

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\dream\AppData\Roaming\npm-cache\_logs\2018-08-01T12_52_17_615Z-debug.log
`

cheton, 2018-08-01:
Please follow the [instruction](https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-production-build) to make production build:

```
$ git clone git@github.com:cncjs/cncjs.git
$ cd cncjs
$ npm install
$ npm run prepare
$ docker build -f Dockerfile -t cncjs/latest .
```

nesquik011, 2018-08-01:
### **i will delete my image docker which i made , and then test yours as testing training  , and see if it works or no , if it did then there is a mistake or file need to be edited because i did try more than 50 times nothing work  **

nesquik011, 2018-08-02:
**### @cheton   i deleted everything as i mentioned before , then i downloaded docker , then pulled docker pull nesquik011/testing1   , then i used this code npm install -g cncjs    , then used this code  start cnc , the server worked , but the changes i made didnt appear , so i am worried about this code npm install -g cncjs which made docker install yours not mine .

so the question here will npm install -g cncjs automatically install your version ? not the one i pulled ?**

nesquik011, 2018-08-02:
or did i miss a step ?

cheton, 2019-04-18:
The build instructions on Windows is now available on wiki

https://github.com/cncjs/cncjs/wiki/Building-Instructions-on-Windows

closed 2019-04-18 by cheton