cncjs/cncjs #374: request new idea

nesquik011, 2018-08-13:
hello sir 

Laser Galvo Mirror check the link  they are used for 3d printers , and not wildly used coz of price and i think 99 % dont know how to control them , so what about you add some option allow us to control them + stl slicer section :D 

https://www.aliexpress.com/item/30K-Laser-Galvo-Mirror-w-Power-supply-DMX-program-board-IC-Animation-laser-light/32407814027.html?spm=2114.search0104.3.8.222b170akRP2nt&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10344_10068_10342_10343_10340_10059_10341_5012515_10696_100031_5012615_10084_10083_10103_10618_10307_10624_10623_10622_10621_10620,searchweb201603_51,ppcSwitch_3&algo_expid=3ae1b459-9fff-48fc-820f-239cc4f263c6-1&algo_pvid=3ae1b459-9fff-48fc-820f-239cc4f263c6&transAbTest=ae803_2&priceBeautifyAB=0

AustinSaintAubin, 2018-08-13:
I think that slicing goes beyond the scope of CNCjs.
I think that the focus right now is to flush-out / improve existing features.
Slicing is a very complicated feature to ask for of a CNC tool mostly aimed at subtractive process work.

It also seems like your request is made for something a motion controller would fulfill, like [Marlin](http://marlinfw.org/).

justinclift, 2018-08-22:
@cheton This issue can probably be closed, as it seems not a CNCjs thing. :smile:

closed 2018-08-23 by AustinSaintAubin