cncjs/cncjs #378: Dev build fails...

benlaurie, 2018-08-30:
Following the directions in CONTRIBUTING, I cloned then did `npm install`. It fails with:

```
> eslint --ext .js --ext .jsx *.js src test

/Users/benl/git-work/cncjs/node_modules/eslint-config-airbnb/rules/react.js:
	Configuration for rule "react/jsx-no-bind" is invalid:
	Value {"ignoreRefs":true,"allowArrowFunctions":true,"allowFunctions":false,"allowBind":false,"ignoreDOMComponents":true} should NOT have additional properties.

Referenced from: /Users/benl/git-work/cncjs/node_modules/eslint-config-airbnb/index.js
Referenced from: /Users/benl/git-work/cncjs/node_modules/eslint-config-trendmicro/index.js
Referenced from: /Users/benl/git-work/cncjs/.eslintrc.js
Error: /Users/benl/git-work/cncjs/node_modules/eslint-config-airbnb/rules/react.js:
	Configuration for rule "react/jsx-no-bind" is invalid:
	Value {"ignoreRefs":true,"allowArrowFunctions":true,"allowFunctions":false,"allowBind":false,"ignoreDOMComponents":true} should NOT have additional properties.

Referenced from: /Users/benl/git-work/cncjs/node_modules/eslint-config-airbnb/index.js
Referenced from: /Users/benl/git-work/cncjs/node_modules/eslint-config-trendmicro/index.js
Referenced from: /Users/benl/git-work/cncjs/.eslintrc.js
    at validateRuleOptions (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-validator.js:133:19)
    at Object.keys.forEach.id (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-validator.js:176:9)
    at Array.forEach (native)
    at validateRules (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-validator.js:175:30)
    at Object.validate (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-validator.js:262:5)
    at loadFromDisk (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-file.js:521:19)
    at load (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-file.js:564:20)
    at configExtends.reduceRight.e (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-file.js:430:36)
    at Array.reduceRight (native)
    at applyExtends (/Users/benl/git-work/cncjs/node_modules/eslint/lib/config/config-file.js:408:26)
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! cncjs@1.9.16-pre eslint: `eslint --ext .js --ext .jsx *.js src test`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the cncjs@1.9.16-pre eslint script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

justinclift, 2018-08-30:
> /Users/benl/git-work/cncjs/node_modules/eslint-config-airbnb/rules/react.js:
>	Configuration for rule "react/jsx-no-bind" is invalid:
>	Value {"ignoreRefs":true,"allowArrowFunctions":true,"allowFunctions":false,"allowBind":false,"ignoreDOMComponents":true} should NOT have additional properties.

Sounds like a mismatch somewhere in the dependency chain, but that's pure guesswork on my part.  Not a node/js person. :smile:

TobyGWilliams, 2018-09-03:
The same error is being talked about here: https://github.com/airbnb/javascript/issues/1893

Running npm ls indicates there's some work to do to update and resolve the dependency tree. 

I will experiment with adding eslint-config-airbnb via npm

TobyGWilliams, 2018-09-03:
Fixed by updating ESlint-plugin-react to 7.11.1.

Pull request created: https://github.com/cncjs/cncjs/pull/380

cheton, 2018-09-27:
Fixed in the master branch

closed 2018-09-27 by cheton