cncjs/cncjs #379: Black webcam screen.

mrp192, 2018-08-31:
#### Description

Hello, Sorry if this is in the wrong spot. I have a logitech C290 webcam but  widget windows is just black even thought it is turned on. I followed the MJPEG-Streamer Install & Setup but that did not help. Again sorry if this is in the wrong spot. 

Thanks for the help.
Brent
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

mrp192, 2018-08-31:
My Pi does see the webcam. Bus 001 Device 004: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
The leds around the webcam are even blue. Just no image on cncjs.

saxa, 2018-08-31:
Can you confirm your webcam works with some other program like Cheese or Gimp or via webrtc ?
Check if your system have the correct drivers loaded for your webcam.

mrp192, 2018-08-31:
Cheese says video error. So now I need to figure out to do drivers.

mrp192, 2018-08-31:
I have also tried Guvcview and I just get an error "could not start video steam in the device" It does list the device.

saxa, 2018-08-31:
Check if you have the correct permissions to access the webcam, you should be part of the correct group. This means is not a cncjs problem.

mrp192, 2018-08-31:
How do I check permissions?

mrp192, 2018-08-31:
alright. So I formatted the CD card and reinstall raspbian. I have made sure the webcam works with cheese. So hopefully everything ill work now.

mrp192, 2018-08-31:
This is still all I get. 
[image]

mrp192, 2018-08-31:
I'm new to this. How would I do that. 

Thanks for the help by the way.

saxa, 2018-08-31:
This is a bug tracker, so I think this is not a cncjs problem. Use google to find out how can you do such things. 
Launch a browser and append the query string parameter ?log_level=debug to turn on debug output.
http://lmgtfy.com/?q=cncjs+debug

mrp192, 2018-08-31:
This is what i'm getting. 

2018-08-31T23:53:56.242Z  DEBUG Loading resource: url="/i18n/en/controller.json" parse@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:691686 vendor.5ece08274c9d5cac7390.js:1
 2018-08-31T23:53:56.244Z  DEBUG Loading resource: url="/i18n/en/gcode.json" parse@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:691686 vendor.5ece08274c9d5cac7390.js:1
 2018-08-31T23:53:56.248Z  DEBUG Loading resource: url="/i18n/en/resource.json" parse@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:691686 vendor.5ece08274c9d5cac7390.js:1
 2018-08-31T23:53:56.435Z  DEBUG Create and establish a WebSocket connection 1684/</</<@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:7427 vendor.5ece08274c9d5cac7390.js:1
 2018-08-31T23:53:56.486Z  INFO CNCjs 1.9.15 e/<@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:7714 vendor.5ece08274c9d5cac7390.js:1
THREE.WebGLRenderer 90 vendor.5ece08274c9d5cac7390.js:1
 2018-08-31T23:53:57.098Z  DEBUG Received a list of serial ports: 
Array [ {…} ]
 serialport:list@http://192.168.0.204/4f6c4342/app.93a4b607d53b7857fb1d.bundle.js?_=1520353491444:1:163635 vendor.5ece08274c9d5cac7390.js:1
THREE.MeshBasicMaterial: .shading has been removed. Use the boolean .flatShading instead.
vendor.5ece08274c9d5cac7390.js:1
navigator.mozGetUserMedia has been replaced by navigator.mediaDevices.getUserMedia app.93a4b607d53b7857fb1d.bundle.js:1
[Show/hide message details.] GET
http://192.168.0.204/4f6c4342/app.css?2ae28e6f18550d5d307431c315b56226
[HTTP/1.1 200 OK 0ms]
Unknown property ‘user-select’.  Declaration dropped. 192.168.0.204:21:15
Expected ‘none’, URL, or filter function but found ‘alpha(’.  Error in parsing value for ‘filter’.  Declaration dropped. 192.168.0.204:51:11
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:72:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:73:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:98:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:99:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:119:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:120:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:130:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:131:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:162:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:163:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:183:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:184:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:194:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:195:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:222:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:223:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:243:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:244:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:254:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:255:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:271:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:272:21
Expected ‘none’, URL, or filter function but found ‘alpha(’.  Error in parsing value for ‘filter’.  Declaration dropped. 192.168.0.204:314:11
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:332:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:333:21
Unknown property ‘zoom’.  Declaration dropped. 192.168.0.204:488:8
Unknown property ‘zoom’.  Declaration dropped. 192.168.0.204:551:8
Unknown property ‘border-top-radius’.  Declaration dropped. 192.168.0.204:575:21
Unknown property ‘border-bottom-radius’.  Declaration dropped. 192.168.0.204:576:24
Unknown property ‘border-top-radius’.  Declaration dropped. 192.168.0.204:579:21
Unknown property ‘border-bottom-radius’.  Declaration dropped. 192.168.0.204:580:24
Unknown property ‘border-bottom-radius’.  Declaration dropped. 192.168.0.204:587:24
Unknown property ‘border-top-radius’.  Declaration dropped. 192.168.0.204:590:21
Error in parsing value for ‘-webkit-background-clip’.  Declaration dropped. 192.168.0.204:61:28
Unknown property ‘-moz-background-clip’.  Declaration dropped. 192.168.0.204:62:24
Unknown property ‘zoom’.  Declaration dropped. 192.168.0.204:40:8
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:158:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:159:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:178:21
Error in parsing value for ‘background-image’.  Declaration dropped. 192.168.0.204:179:21
[... cut ...]
Unknown property ‘line-break’.  Declaration dropped. 192.168.0.204:200:14
Expected ‘none’, URL, or filter function but found ‘alpha(’.  Error in parsing value for ‘filter’.  Declaration dropped. 192.168.0.204:215:11
Expected ‘none’, URL, or filter function but found ‘alpha(’.  Error in parsing value for ‘filter’.  Declaration dropped. 192.168.0.204:336:11
Error while fetching an original source: TypeError: NetworkError when attempting to fetch resource.
Source URL: webpack:///app.93a4b607d53b7857fb1d.bundle.js[Learn More]

closed 2018-09-01 by mrp192