cncjs/cncjs #381: Development tools - front end testing

TobyGWilliams, 2018-09-04:
Hi,

I've been thinking maybe cncjs should have some front end testing tools. I find working on the front end particularly difficult because of this. 

My personal preference is Karma and Jasmine but that's only really due tohaving the option to have the test rendered in the browser.  (I believe, definitely true for my other projects) 

Any thoughts?

Toby

nesquik011, 2018-09-04:
Like what ? I wonder what testing u are talking about

TobyGWilliams, 2018-09-05:
If you are not familiar with TDD then here is a good place to start: https://en.wikipedia.org/wiki/Test-driven_development

MitchBradley, 2018-09-05:
I would be interested in hearing about the scope of what you propose - perhaps a few examples of what might be tested.  Given CNCjs's role as a middleman between a user and one of many controllers, in turn attached to different machines, it would seem that a useful test suite would need to be enormous.
CNCjs does almost nothing until it connects to a controller, so I think a first requirement for automated testing would be a way to simulate a controller or controllers.  Early in my involvement with CNCjs, I tried to use grbl-sim to test basic CNCjs functionality in a cloud build, without much success.  I tried to fix grbl-sim (it emulates an AVR UART at a hardware register level and tries to be timing-accurate, which runs afoul of Linux timing and process switching granularity), but decided it wasn't worth my time.
Once some flavor of controller simulation is available, there will still be a large matrix of controller state-transition and timing possibilities to model in order to test CNCjs well.  With 5 asynchronous elements (machine, controller, CNCjs server, CNCjs browser, user), things get complicated really fast.
But maybe I don't understand the scope of your proposal.

TobyGWilliams, 2018-09-05:
I agree that full end to end testing is near impossible, I'm proposing unit and integration testing for the front end development (CNCjs browser). There is already testing in place via Tap for the controllers on the application side. 

At the moment, if I'm working on adding functionality to the UI, I have to repeatedly manually navigate to the same place in the UI every time to see if my changes have had the desired impact. Similarly, there's nothing to tell if changes have had negative impact on another element. 

Tools to choose from are mainly: Karma/Mocha/Jest + Expect/Jasmine/Chai/Assert.

Jest doesn't appear to have support rendering individual components and I'm personally more familiar with Karma than Mocha.  

I've created a branch to play with testing here: https://github.com/TobyGWilliams/cncjs/tree/Testing-Framework. My progress is limited as I haven't had much luck making webpack and babel play nice with Karma.

Toby

MitchBradley, 2018-09-05:
Interesting.  I should probably do something like this for cncjs-shopfloor-tablet, which I maintain.  It doesn't use webpack or babel so maybe Karma will like it better.  It would be especially helpful for the RPN calculator that comes up for numeric entry.  That has a lot of edge conditions that would benefit from a test suite, with few if any external dependencies.

cheton, 2019-04-20:
With the increased complexity, it's pretty important to have a complete UI testing methodology to make sure new things won't break existing behavior. 

I may choose WebDriverIO (with a Selenium server) or Google Puppeteer for UI testing. I might get it started when I have some free time this year.