cncjs/cncjs #382: OS remote access when running on gui.

f0m3, 2018-09-16:
#### Description

When I start cncjs on OSX from gui and with gui I can't connect to it from a browser. This only works when I start it from terminal without gui. Can I connect to cncjs when it is running with gui? 

#### Versions

- CNCjs: 1.9.16
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [0] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [x ] Mac
- [ ] Linux

cheton, 2018-09-28:
Go to **View** and click **Home** at the bottom, it will open a browser window containing IP address and listen port that allows you access the UI via browser.

[image]

closed 2018-09-28 by cheton

funbobby2001, 2019-04-09:
FYI - I was having the same problem and realized through the above tip that CNCjs (Windows desktop version) was picking a new host IP and port every time it was launched. I modified the shortcut target properties in Windows to run "C:\...\cncjs.exe --host 192.168.1.206 --port 8000" and now it works as advertised. Happy day for me!