cncjs/cncjs #383: Many unnecessary things in the UI

makarenya, 2018-09-18:
#### Description

Hi!
I'm using cncjs, and i found, that it sometimes have a LOT of unnecessary UI items. My CNC have Marlin firmware (yes, it is 3 in 1 homemade device), and, for example, it hasn't any work coordinates system. Really, Marlin have only ONE coordinate system, that can be switched to zero in any point. And Marlin have no coolant Control, but i have coolant buttons... And many many more unused things.

In the same time, the Primary things, that I need to:
- Home all axis (Z first, and then X and Y simultaneously)
- Zero XY plane at current point
- Zero Z axis at current point
This is features, that i need every time, that i use CNC - and I can get it only through second level of menu. (XY simultaneously homing not available at all)...
As result - many things, that don't working on my firmware - I see it all behind my face, and few things, that i really need - i cat reach only from deep menus...

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?
NPM

#### CNC Controller

Marlin

#### Hardware

Raspberry Pi

#### Operating System

Not Applicable

MitchBradley, 2018-09-18:
You might consider one of the simplified UIs like cncjs-shopfloor-tablet .  I wrote it to do exactly what I need with no extra clutter, which sounds like what you are asking for.  Some modifications would be needed to do exactly what you want, but it is easier to modify than the full-blown cncjs because the code is much smaller with fewer dependencies on huge frameworks.

cheton, 2018-09-21:
Hi @makarenya 

You can create MDI buttons to fulfill your needs. It's available since v1.9.13.

[image]

Toggle MDI mode
[image]

closed 2018-10-08 by cheton