cncjs/cncjs #384: GRBL 0.9j proeblem, not activate buttons

luisluna2, 2018-09-18:
CNCjs 1.9.16-pre-896

Buttons of jogginf and spindle Not working when connect

[image]

luisluna2, 2018-09-23:
> Your Grbl controller does not respond correctly. Try disabling "Enable hardware flow control" before connecting, or clicking the Reset button and check the message in the console widget.

Hello friend, I did what he told me, but it does not work.

cheton, 2018-09-25:
@luisluna2 You can connect to Grbl via Arduino serial monitor. If everything is ok, a welcome message will pop up `Grbl 0.9j ['$' for help]`, and you can see status output by sending the "?" command.

[image]

If you cannot make it work via Arduino IDE, try upgrading your Grbl firmware from 0.9j to 1.1f.
https://github.com/gnea/grbl/releases
https://github.com/gnea/grbl/wiki/Compiling-Grbl

luisluna2, 2018-09-25:
@cheton Thank you Friend, I will check this tomorrow and I will reply.

luisluna2, 2018-09-25:
hello check this:

[image]

and same:

[image]

MitchBradley, 2018-09-25:
Okay, it looks like you have limit switches enabled in your GRBL configuration.  There are several possibilities:

1. You actually have limit switches and they are active, i.e. the machine is already at its limit position
2. You have limit switches and they are misconfigured - the polarity is wrong so they are saying the machine is at the limit when it is not
3. You do not have limit switches but your GRBL configuration is set up as if you did have them - so GRBL is trying to sense switches that do not exist and it thinks they are active

You can turn off GRBL limits by typing the following command in the serial monitor window:

`$21=0`

If you do have limit switches but their polarity is wrong, you can change the polarity with

`$5=1`

or 

`$5=0`

$5=0 is for the default behavior where pin low means limit triggered.  $5=1 is for the opposite behavior - pin high is limit triggered.

luisluna2, 2018-09-25:
@MitchBradley I try the 3 options and the same, my GRBL is 0.9j maybe with new 1.1 can work, I'm checking now...

MitchBradley, 2018-09-25:
Do you have soft limits configured?

luisluna2, 2018-09-26:
@MitchBradley  yes friend, but only I have a arduino for check the program, not connect in my CNC, but I check with 1.1f and all is okay!, I will change of GRBL version.