cncjs/cncjs #385: validity of 'cncjs-app-1.9.15-mac-x64.dmg'

core-code, 2018-09-29:
hey there,

when cncjs-app-1.9.15-mac-x64.dmg was first posted, it had the checksum
d62fc40284a03d4a999b3118b77851e81e4eeab1191eb9997a595a576d5ce689

now it suddenly has the checksum
2eebe42010c8b1ececa2591eaeacdd343d7b8d07509841d3ad05e0e36f0ba091

since the file was silently updated without explanation and has no codesigning in-place to confirm its authenticity, its hard to confirm whether it is a valid release or whether it has been tampered with.

could you give any confirmation about the validity of the file?

many thanks

cheton, 2018-09-29:
Hi @core-code 

It's safe to download files from the GitHub releases page since all those files were created from Travis CI and Appveyor, not manually uploaded from personal devices.

The reason of seeing different checksums might be a rebuild triggered on Travis CI for applying urgent bug fixes, and you may accidentally download the previous copy when it was firstly posted.

You can use <b>cncjs-app-1.9.15-mac-x64.dmg</b> with the checksum 2eebe42010c8b1ececa2591eaeacdd343d7b8d07509841d3ad05e0e36f0ba091. It was updated on `2018-07-29T08:15:33Z` according to GitHub release log.

closed 2018-09-29 by cheton

cheton, 2018-09-29:
A new build of 1.9.15 was updated by https://travis-ci.org/cncjs/cncjs/builds/349830708

cheton, 2018-09-29:
@core-code In general, bug fixes and enhancements will be applied to a new version (e.g. v1.9.16). I forgot why it triggered a new build for 1.9.15, it might be caused by merging code with a pull request that follows the v1.9.15 tag.