cncjs/cncjs #388: Watch folder problem

Flupke1979, 2018-10-20:
Hello,
First of all thanks for the great work with the software.

The problem I have is that I cannot get the watch folder working.
When typing the command **cnc --watch-directory /home/pi/watch** I always get the error:

2018-10-20T22:29:11.265Z - info: init Loading configuration from "/home/pi/.cncrc"
2018-10-20T22:29:11.282Z - info: init Watching "/home/pi/watch" for file changes
2018-10-20T22:29:12.262Z - error: init { Error: listen EADDRINUSE 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (net.js:1286:14)
    at listenInCluster (net.js:1334:12)
    at doListen (net.js:1460:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:744:11)
    at startup (internal/bootstrap/node.js:285:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:739:3)
  errno: 'EADDRINUSE',
  code: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }
Error: { Error: listen EADDRINUSE 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (net.js:1286:14)
    at listenInCluster (net.js:1334:12)
    at doListen (net.js:1460:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:744:11)
    at startup (internal/bootstrap/node.js:285:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:739:3)
  errno: 'EADDRINUSE',
  code: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }

Any thougths or ideas?

Kind Regards,
Flupke

AustinSaintAubin, 2018-10-20:
Hi @Flupke1979 
The following like has a good indicator `EADDRINUSE`
> 2018-10-20T22:29:12.262Z - error: init { Error: listen EADDRINUSE 0.0.0.0:8000

Its likly that the [Address is already in use](https://stackoverflow.com/questions/4075287/node-express-eaddrinuse-address-already-in-use-kill-server).

Run `ps aux | grep node` or use `htop` to see if CNCjs is already running.
You can also [see whats listening on that port](https://www.cyberciti.biz/faq/what-process-has-open-linux-port) `netstat -tulpn | grep :8000`

Flupke1979, 2018-10-20:
Hello Austin,
Thanks for the quick reply.
CNCjs is running.
According to netstat it is cncjs that is listening on the port.
I stopped it with PM2 stop cncjs
(Do not know if it is the right way)
Then did the command again and got the message:
2018-10-20T23:10:03.257Z - info: init Loading configuration from "/home/pi/.cncrc"
2018-10-20T23:10:03.289Z - info: init Watching "/home/pi/watch" for file changes
2018-10-20T23:10:04.362Z - info: init Starting the server at http://127.0.1.1:8000
2018-10-20T23:10:09.191Z - warn: service:cncengine Error: Unauthorized user: user.id=, user.name=cncjs-pendant
    at /usr/lib/node_modules/cncjs/dist/cnc/app/index.js:6933:20
    at new Promise (<anonymous>)
    at new F (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:479:28)
    at validateUser (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:6911:12)
    at CNCEngine._callee$ (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:16695:72)
    at tryCatch (/usr/lib/node_modules/cncjs/node_modules/regenerator-runtime/runtime.js:62:40)
    at Generator.invoke [as _invoke] (/usr/lib/node_modules/cncjs/node_modules/regenerator-runtime/runtime.js:296:22)
    at Generator.prototype.(anonymous function) [as next] (/usr/lib/node_modules/cncjs/node_modules/regenerator-runtime/runtime.js:114:21)
    at step (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:1363:30)
    at /usr/lib/node_modules/cncjs/dist/cnc/app/index.js:1374:13
    at process._tickCallback (internal/process/next_tick.js:68:7)

AustinSaintAubin, 2018-10-21:
After reviewing your issue. Looks like you just want to add the watch-directory flag to the run statment.
One way to do this is:
```
pm2 stop $(which cncjs)
pm2 start $(which cncjs) -- --port 8000 -m /tinyweb:/home/pi/tinyweb --watch-directory /home/pi/watch
```

Flupke1979, 2018-10-21:
It works. Thanks.

closed 2018-10-21 by Flupke1979