cncjs/cncjs #389: A-Axis does not report degrees.

toddmitchell, 2018-10-25:
#### Description
Hi There,

I am using a rotary axis with my cnc machine and the A-axis in the axis widget does not report/display a value when jogging (i.e. G0A250).  I did not try the G1 command.

I am configured to use millimeters on a cohesion3d board (smoothieware) and custom compiled firmware following smootheiware's docs with 5 axes enabled.

Glancing through the code, I didn't see any conversions to degrees but I could be missing something?

Thanks!
#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2019-04-23:
https://github.com/Smoothieware/Smoothieware/blob/047da095df0ce1ccd7f73baf0e2fc1c547b6515e/src/libs/Kernel.cpp#L163

It seems the status string has changed over time and now in different format. What will it report when you enter the "?" command or the "get status" command over serial console?

gitwasi, 2022-04-03:
I have the same with Marlin 2.0.9.3   (CNCjs 1.9.24)

> client> M114
> X:6.000 Y:606.000 Z:409.000 A:10.000 B:-83.000 Count X:2793 Y:279669 Z:163600 A:158 B:-1308
> S_XYZ:5
> ok

but values for A and B axes on Axes tab are 0.000

cheton, 2022-04-03:
Hi @gitwasi,

Rotational axes can be supported by adding 'A', 'B', 'C' to the parser.

Original linear X/Y/Z axes:
```js
// https://github.com/cncjs/cncjs/blob/v1.9.24/src/server/controllers/Marlin/MarlinLineParserResultPosition.js
class MarlinLineParserResultPosition {
  // X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
  static parse(line) {
    const r = line.match(/^(?:(?:X|Y|Z|E):[0-9\.\-]+\s+)+/i);
        :    :    :
    const pattern = /((X|Y|Z|E):[0-9\.\-]+)+/gi;
```

Adding rotational A/B/C axes:
```js
class MarlinLineParserResultPosition {
  // X:0.00 Y:0.00 Z:0.00 A:0.00 B:0.00 C:0.00 Count X:0 Y:0 Z:0 A:0.00 B:0.00 C:0.00
  static parse(line) {
    const r = line.match(/^(?:(?:X|Y|Z|A|B|C|E):[0-9\.\-]+\s+)+/i);
        :    :    :
    const pattern = /((X|Y|Z|A|B|C|E):[0-9\.\-]+)+/gi;
```

I can create a PR to support more axes for the Marlin controller, the only problem is that I don't have a Marlin firmware configured with 4 or more axes on hand.

A quick way to verify the change is to patch existing code with above axes. Could you tell me your CNCjs running environment (e.g. Linux, Windows, macOS apps, or installed via NPM)?

gitwasi, 2022-04-03:
I use this now on Windows x64 desktop app and I can also test it on Linux desktop. I can configure all 6 axes on Marlin for testing.

gitwasi, 2022-04-03:
I've done @cheton Your changes locally in the code, made .AppImage from this and now it is working perfect for all 6 axes.
It is awesome to look at this when all axes move and this numbers are changing.

for 6 axes on Marlin 2.0.9.3 M114 give this:
> X:20.000 Y:41.000 Z:38.000 A:34.000 B:24.000 C:17.000 Count X:9311 Y:18922 Z:15200 A:536 B:378 C:268

I didn't tested that on windows because I don't know how to make windows binaries.

off topic but:
It would be great to do something with this "S_XYZ:...". When in Marlin configuration continuous reporting is enabled (REALTIME_REPORTING_COMMANDS) that code is filling out all the console.