cncjs/cncjs #390: Gcode File Name

nickm324, 2018-10-27:
#### Description

Gcode file name in 3D display is to small to even read.

#### Versions

- CNCjs: 1.9.15

[image]

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2018-10-31:
I agree. I think it would be better / nice to have the file name displayed at the top of CNCjs.
Any overflowing text could be clipped, hovering over the clipped text could show tooltip that displays the entire filename.
[image]

cheton, 2019-03-16:
Fixed in 1.9.17

closed 2019-03-16 by cheton