cncjs/cncjs #391: MJPEG Refuses to Stream

AcrimoniousMirth, 2018-10-31:
Trying to stream webcam via RPi. Followed steps in guide and in FAQ, checked that the camera shows up, video0 is there and can even view footage natively on RPi. Managed to get it to stream for a while after following the FAQ steps, then restarted the RPi to check if it'd continue working and it didn't. Attempted to restart the stream, claimed it already was (suggesting auto startup works), but no image. Checked MJPEG settings and tried again, still no biscuit.

- CNCjs: 1.9.15

Installed onto RPi as instructed on Wiki.

- [x] Grbl

- [x] Raspberry Pi

- [x] Not Applicable

AustinSaintAubin, 2018-10-31:
Did you follow the [section on Startup](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-MJPEG-Streamer-Install-&-Setup-&-FFMpeg-Recording#start-on-boot) for the Video Stream manger script?
Also, have a look at the [video scripts that I use for the CNCjs Raspberry Pi image](https://github.com/cncjs/cncjs-pi-raspbian/tree/master/Videos).
I recommend using htop to see what processes are running when you first boot, you can filter for you stream command to see that / if it is running.

```
# Install htop
sudo apt-get install htop

# Run htop
htop
```

You can also use Cheese to test, it will let you take a single photo. I think that is what I used early on. 
```
# Install & Use Cheese
sudo apt-get install cheese

# Cheese Help
cheese -h
```

AcrimoniousMirth, 2018-11-01:
@AustinSaintAubin 
Hi Austin! Yes, I did. I noticed it did seem to be operating at startup, just wasn't connecting through somehow. I'll take a look at htop when I have a chance! I used fswebcam to test the camera itself.
Will get back to you when I've tried htop.

AustinSaintAubin, 2019-05-16:
@AcrimoniousMirth 
You have a chance to test? Is this still an issue?

AustinSaintAubin, 2019-08-23:
Hi @AcrimoniousMirth you still having this issue?

AcrimoniousMirth, 2019-08-23:
Hi @AustinSaintAubin 
I haven't been very involved since my last comment so can't contribute anything useful here. Moved country, didn't have a chance to take my machine with me. 
I'm going to close the issue.

closed 2019-08-23 by AcrimoniousMirth