cncjs/cncjs #393: No port available from localhost

dpichevin, 2018-11-18:
#### Description

I have installed CNCjs on a Raspberry Pi 3, with a fresh Raspbian image. I can connect to it from other machines or my tablet and use the CNC, but when I try to use it directly from the server in Chromium, there is no port available to connect. The reason I want to use it from the same machine is that it's all in the garage and it looks it's too far away from the wifi router and the speed is poor, every command lags and it's not really usable (for example jogging). I want to see if it's better when used locally. Any ideas?

Thank you.

#### Versions

- CNCjs: 1.9.15
- Node.js: 11.2.0
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2018-11-19:
Take a look at the [CNCjs Raspberry Pi pre-built image](https://github.com/cncjs/cncjs-pi-raspbian/releases).
This image has a Kiosk mode that start directly into a web interface.
This web interface is set to a camera feed by default.
This is because of limited OpenGL support on the Pi, need for rendering the preview.

SSH into the the Pi after installing the image and alter `/home/pi/.config/openbox/autostart`, removing the '.../tinyweb' from the web address.
`nano /home/pi/.config/openbox/autostart`
then change, the Start Chromium in kiosk mode.
```
# Start Chromium in kiosk mode
chromium-browser --noerrdialogs --disable-suggestions-service --disable-translate --disable-save-password-bubble --disable-session-crashed-bubble --disable-infobars --touch-events=enabled --disable-gesture-typing --kiosk 'http://localhost:8000'
```
Save and reboot

Also, consider wired networking the Raspberry Pi (not WiFi).

dpichevin, 2018-11-19:
Thanks, I'm not sure I understand your answer. Are you telling me that installing that image would solve the issue of not having a port available locally even though that port is available when connecting via another machine? 

I cannot network the Pi, it's wireless only, my garage is not attached to the house.

dpichevin, 2018-11-19:
OK, the wired network did the trick. I was able to ssh, setup wireless, and make the changes above. Also, had to remove -nocursor from the startx command.

Unfortunately, the net result is still the same as before, there is no port available from the Pi (now in kiosk mode). I can connect to it just fine from my PC and select the port with the CNC machine on it,

AustinSaintAubin, 2018-11-19:
Post some video and I will work with you on this.
The Raspberry Pi is a little under powered for intensive work. That could account for the lag. But it could be other things

dpichevin, 2018-11-19:
Thanks. I'm still in the setup phase so far, so I cannot tell if there will be an actual issue when cutting. We'll see :) I also bought an older laptop when I thought I would use a more resource demanding app like Carbide Motion, then discovered CNCjs and decided to go back to my original idea of using a Pi I already had, which is one I can easily leave in the garage.

MitchBradley, 2018-11-20:
I have been underwhelmed with the cncjs GUI performance of a Pi, but the Pi works great as the server component, with the GUI running on a $100 Android tablet, especially with the cncjs-shopfloor-tablet simplified UI.  I do my production milling with Arduino Due/g2core + Pi Zero W/cncjs server + cheap Android tablet / cncjs-shopfloor-tablet UI.  That UI strips out all the stuff you don't need to run jobs in the shop, in favor of big, hard-to-mishit buttons for the stuff you use all the time.  You can still switch to a full-cncjs browser tab for the infrequent stuff, but I find that I never need to do that anymore.

closed 2019-03-07 by cheton