cncjs/cncjs #394: CNCjs seems to fall asleep

eciramella, 2018-11-20:
#### Description

After following all the setup steps on the wiki, I've made a few successful cuts with CNCjs.  I usually have a macbook around somewhere, and watch the main CNCjs UI to speed up/slow down feed rates.

Recently, the CNCjs UI seems to freeze and fails to show progress on the tool head (none of the buttons seem to work at this point).  To fix, I have to refresh the browser tab I have open for CNCjs.  MOST times, it comes back.  But sometimes, I need to restart my PI, re-home, etc.

CPU/Memory seems ok/low, temp of the pi seems reasonable.  I see nothing in the log setup by PM2 (I see other events, but there aren't any errors).

#### Versions

- CNCjs: 1.9.15
- Node.js: 8.12.0
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x] Not Applicable

rbenji23, 2019-01-02:
Having the same issue on the same version 1.9.15. I am using the prebuilt mac version. Usually if I pause and then hit run it starts working again. Happens about every 20mins of run time.

markbdusted, 2019-01-04:
I have the same issue. The carve will finish but the UI is frozen and it only shows about half the lines being sent. At this point none of the buttons work. 
The only resolve I can find is closing the connection and re opening. 
Using same version as the OP on a windows 10 PC.

AustinSaintAubin, 2019-01-06:
Odd, open the [Developer Tools](https://developers.google.com/web/tools/chrome-devtools/) and ( Control+Shift+J ) and see what errors are reported.

markbdusted, 2019-01-10:
I later noticed if quickly flashes a message about server failing.The notice is only visible for about 1/2 a second. 
I will try to open the developer tools the next time it happens. Are the reported errors saved between uses?
Because when it happens I don't think I will be able to do anything without reloading.

markbdusted, 2019-01-13:
When this happens the program is dead and I can do nothing and have to
reload. Even though the carve finishes.
I thought maybe the 3d view was too much demand so I tried with it turned
off and the same thing happened.

markbdusted, 2019-02-09:
Have you tried to reload while the machine is running code?
I it kind of a PIA and a delay in work flow waiting until after carve is complete. Then you have to wait for it to reload.

TNTcreations, 2020-04-14:
I'm running 1.9.22 on Windows 10.  I have noticed that the Grbl Widget will stop reporting the spindle speed and will show that the spindle is off during the cut.  After awhile, it does show that the spindle starts again.  I will be adding a spindle soon, so I will update if it actually affects the spindle or if it is just not reporting.

TheSilentEngineer, 2020-04-14:
Running on
MacOS Catalina
CNCjs 1.19.21 / 1.19.22 
Grbl
Electron install.

Same general issues, Axis display stops updating and console stops updating. Also can't type commands into console, or at least there is no update to the display. This happens randomly and it doesn't seem to effect operation. I must close and reopen the connection to reset it.

vallettea, 2020-07-23:
Same error on raspberry pi but I can't find a way to resume the operation.

nickm324, 2020-07-23:
I reported this same issue over a year ago, if you go back to version .13 it will no longer be a problem. Unfortunately I could not find a resolution after many days of trouble shooting and in the end resorted to installing everything on an old PC I had that is running Ubuntu. Haven't had an issue since.

rhass, 2020-08-14:
@nickm324 Your comment makes me wonder if the wifi power save mode on the RPi and other devices is a contributing factor. The WiFi Power Management on the RPi (4) is very aggressive and causes some pretty bad latency if the wifi is idle more than about two or three seconds. 

Try `sudo iw wlan0 set power_save off` and see if that helps at all for this issue.

nickm324, 2020-08-14:
I had already tried disabling power save mode and it made no difference.

emcniece, 2022-03-26:
Hey, I know this is a tad late, but does anyone recall what kind of Gcode filesizes are being processed when this issue happens?

markbdusted, 2022-03-26:
I do not recall and have changed to another sender.
Sorry

benjaffe, 2022-04-15:
I've experienced this with pretty small files, like 2kb. I've also noticed it after running a cut, and then trying to run another, but that might be more coincidence than part of the problem.