cncjs/cncjs #397: macros don't work with gcode

nesetburak, 2018-11-28:
#### Description

How can ı use macros with g code. ı am **UPLOAD** this gcode;
`G21 G90 `
`G01 X10 `
`T1 M6`
spindle move to X10 coordinate but don't start ''T1 M6'' macros 

[image]

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2018-11-28:
CNCjs macros are separate from GCode macros.
In the GCode language, macros are words starting with the letter M. Their behavior is either standard or defined by the designer of the machine controller hardware and firmware. They are implemented by the machine controller.  The machine controller is the microcomputer running GRBL, TinyG, Marlin, or Smoothieware, so GCode "M" macros are handled by the GRBL (etc) firmware.  The firmware runs a macro when it receives the "M" code word in the GCode stream.  GCode macros do not have to composed of sequences of other GCode words; in fact, they are usually not implemented that way, because they often perform machine control functions that cannot be done by any other GCode word. 
 The word "macro" is thus a bad choice, because in the general computing community, "macro" usually means "a command composed of a sequence of other standard commands".  But the GCode language has used the word "macro" for a long time and it is now part of the language definition. Users cannot change GCode "M" macros, except by modifying GRBL source code, recompiling, and installing the new GRBL firmware image on the Arduino.
CNCjs has an additional layer of macros that is separate from GCode macros.  CNCjs macro names can be arbitrary text strings.  If you choose to use the name "M6", that looks like a GCode macro name, but it is still a CNCjs macro that is separate from the GCode macro names.  CNCjs macros are executed interactively by the user, by clicking on the "run" arrow to the left of the name.  They are not executed from a stream of GCode text.  CNCjs macros are user-defined.  They are truely macros in the general computing sense, since they are composed of sequences of GCode commands, with some additional syntax for inserting parameters.
I think that there have been some requests for the feature that you want - the ability to run user-defined code sequences from GCode programs.  It would indeed be a useful feature.  At first glance, it seems like it would be an easy thing to do, but when you dig deeper it becomes very tricky.  The GCode language is very poorly designed with lots of modes that change the behavior of code sequences.  Controllers like GRBL have strict requirements for flow control, and multiple levels of buffering that make it hard to precisely synchronize the machine state, controller state, and user interface (CNCjs) state.  GCode synchronization, flow control, and error reporting often boil down to counting lines - and any form of GCode-triggered user macros deeply affects the line numbering scheme. If you did have a simple form of the feature, that would instantly trigger a flood of requests for extensions to make it into a proper programming language - things like subroutines, looping, variables, expressions, conditionals, etc.
Some standalone GCode systems do have such programming features. For example, the GCode language has subprograms and parameters. Experience has shown that they are very hard to use because the GCode language is poorly designed.  It grew up in an ad hoc fashion from a very simple language with just a few commands, and extensions were added by different people, usually without any knowledge of real computer language design. Controller firmware like GRBL does not implement the extensions, partially because the Arduinos that run GRBL don't have enough memory, and partly because people rarely use the programming features anymore.  Instead of the programming features, CAM software is used to regenerate whole GCode programs using only the simple straightforward features.

nesetburak, 2018-11-28:
firstly thank you for this gathered answer. ı learned about gcode history. ''macros'' terminology were so interesting. 

you say;
if ı want to solved this problem, ı'll open grbl or g2core firmware and modify it.
for example;

[image]

MitchBradley, 2018-11-28:
If your intent is to perform a tool change - which is the usual meaning of "T1 M6", please see this page:

https://github.com/cncjs/cncjs/wiki/Tool-Change

CNCjs will stop to let you manually run your CNCjs macro.

MitchBradley, 2018-11-28:
Here are some relevant discussion threads:

This one triggered the existing tool change functionality in CNCjs:
https://github.com/cncjs/cncjs/issues/118

which resulted in this issue being raised on GRBL:
https://github.com/gnea/grbl/issues/91

the result was that GRBL was changed to keep track of and report the tool number, but that is the extent of GRBL's support for tools.  No tool tables or tool offsets.
https://github.com/gnea/grbl/blob/master/grbl/gcode.c#L315

g2core supports tool tables with tool offsets:
https://github.com/synthetos/g2/wiki/Tool-Offsets-and-Selection

CNCjs pauses on M6 for all controller types.

closed 2019-03-07 by cheton