cncjs/cncjs #398: Run macros from pendant

Grindel, 2018-12-05:
#### Description

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

darickc, 2022-01-12:
I know this is pretty old, but were you able to figure this out?  I am trying to do the same thing.  I thought that calling "macro:run" would do the job, but when I do that, none of the variables are filled like it is when I run it from the main UI.

darickc, 2022-01-13:
The issue is that when you run a macro from a pendant it doesn't use the context (the min/max) of the loaded gcode in the main UI.  So you have to pass context with the macro:run command.  Unfortunately, I don't see the context from the main UI exposed anywhere (there is a context in the sender status, but it is empty, I haven't looked at the code to see what this context is referring to) so you have to build the context based on the loaded gcode yourself to pass with the macro. Hope this helps someone else.

vertfin, 2023-03-05:
Using darickc's comment for guidance, I was able to get the cncjs-ky-ext macro running from my pendant (https://github.com/vertfin/cncjs-pendant-numpad).  When gcode is loaded, the gcode's bounding box is computed (via code from https://github.com/cncjs/cncjs-shopfloor-tablet).  When the macro is run, a context is built using things from the current controller state and bounding box.  I was surprised I couldn't find an existing npm package to get the bounding box.  There may be a need for a "gcode-statistics" package.

Billiam, 2023-03-05:
A bounding box calculator would be really helpful.

@vertfin  Note that the shopfloor tablet bounding box calculation will be [incorrect for some gcode moves](https://github.com/cncjs/cncjs-shopfloor-tablet/issues/36), which can be misleading if you use the bounding box for anything more serious than gcode previewing.

roblesterjr04, 2025-06-20:
@vertfin Can you share some code or more details? I have a heavily customized UI for my shopfloor UI and i'm trying to call the autolevel macro, and running in to the same issue with no bounding context. What i'm not sure of is how to pass that context to the macro, and where to pull the bounds from. This is my line of code

```js
cnc.controller.command('macro:run', '84fa008c-f3bc-4819-96a7-6ed617a04330');
```

Also, do I need to run `macro:load` before this? Thanks!

roblesterjr04, 2025-06-20:
@Billiam it must be though, because the shopflooor UI has a 2D drawing representative of the bounding box. I'll have to dig.

Billiam, 2025-06-20:
@roblesterjr04 The shopfloor project uses its own simplified code for parsing the gcode and displaying the bounding box: 

https://github.com/cncjs/cncjs-shopfloor-tablet/blob/a44560bce2191dd087c2436b48a0d348ac7f34cc/src/toolpath-displayer.js#L167-L292

vertfin, 2025-06-22:
> [@vertfin](https://github.com/vertfin) Can you share some code or more details?

You can check my pendant code at https://github.com/vertfin/cncjs-pendant-numpad specially the gcode_bbox.js file.  I think I borrowed that from https://github.com/cncjs/cncjs-shopfloor-tablet but it's been a while.  I just wanted a button on my pendant that moves the machine around the cutting perimeter.  So far it's worked fine for my purposes (cutting pcbs on a 3018).  I hope this helps!