cncjs/cncjs #401: Remove tooltip from custom buttons

nickm324, 2019-01-02:
#### Description
** IMPROVEMENT**
Remove the tooltip from custom buttons setup by the user.  if you have a long list of gcode commands assigned to the button the tooltip that is displayed is very large and covers a lot of the interface. A tool tip here is not needed since this is assigned code to the button and the user is aware of the underlying commands already.

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.15

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

nickm324, 2019-03-10:
Was this addressed in the 1.9.16 update?

cheton, 2019-03-11:
@nickm324 
Sorry for that I forgot adding the commit to 1.9.16.
I will make the change in 1.9.17, it will be probably released in this week.

nickm324, 2019-03-11:
Cool, thanks.

cheton, 2019-03-16:
Fixed in 1.9.17

closed 2019-03-16 by cheton