cncjs/cncjs #402: Android WebApp Capable [Improvement]

nrepian, 2019-01-10:
#### Description

When viewing CNCjs on an android device (for example, using it as a pendant), there is a lot of screen space wasted by the address bar. It's possible to remove this when adding the CNCjs URL the Android Home screen, as it should be able to launch as a web app. It requires a short amount of code to be added to the header to enable this:

<meta name="apple-mobile-web-app-capable" content="yes">

Source: https://developers.google.com/web/fundamentals/native-hardware/fullscreen/

It's a minor tweak that won't affect anyone other than users using this as a webapp, as I imagine many of us do.

#### Versions

- CNCjs: 1.9.15

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-01-29:
Take notes:
https://developer.apple.com/library/archive/documentation/AppleApplications/Reference/SafariHTMLRef/Articles/MetaTags.html

karoria, 2019-03-26:
I have given it a try with cncjs shopfloor pendant. Technically it goes full screen but the layout changes drastically as the size of buttons and text fields are relative I think. But if somebody can workout to freeze the same layout in fullscreen, it will be a huge advantage working in full screen on android and apple phones.

MitchBradley, 2019-03-26:
When you say "Android device", are you talking about a tablet or a phone or both?  The shopfloor tablet screen layout works pretty well on a 10" tablet, but it is not setup for a phone-sized screen.  You might be able to do something with the .css to make a phone-centric layout, but I think it would require rethinking the overall layout instead of just scaling and freezing.  I have been thinking about structuring the UI as a set of task-focused subscreens, instead of the current "one screen to rule them all" approach.  There would be different screens for job setup/jogging, task running, and maintenance.  Each screen would have only a few controls, with large buttons.

MitchBradley, 2019-03-26:
Update: I just tried cncjs-shopfloor-tablet on an iPhone.  It is sort of okay in landscape mode, but the gcode section at the bottom is off-screen and you have to scroll to get to it.  This is an example where the subscreen approach might work - the gcode section could be moved off to a task-run screen.
In portrait mode, the buttons are just too small.

karoria, 2019-03-27:
Wonderful idea of task focussed subscreens @MitchBradley. This will also give space to some more buttons and info display like spindle on/off, alarm description and folder view for program files if possible. BTW, these are just suggessions from my few days of experience with RasPi 7" touchscreen. I am eagerly waiting for this pendant.

MitchBradley, 2019-03-27:
If you want to try your hand at designing subscreens, make a set of drawings showing proposed button layouts.  You can use any drawing program, or just scribble on paper and take a photo.  Design and coding are two different tasks.  You could contribute to the first even if you aren't a coder.

Here are a couple of design guidelines:

- Things should be laid out in some sort of grid, because that makes coding much much simpler.  The "bootstrap" framework that we use divides the screen into 12 columns and each element can occupy any number of those columns, with the total number adding up to 12 on each row.  So you could have 4 buttons across, each occupying 3 columns, or 2,2,2,3,3, or any arrangement that adds up to 12 across.
- I try to keep a good separation between things that cause motion - or at least make the buttons big - to avoid mis-hits.
- Things that are not relevant, or that cannot be used, during a given operation or state are disabled (grayed out). 

BTW, my latest version of cncjs-shopfloor-tablet has spindle speed and feedrate info - overlaying the jog button area when running a job.  It has spindle-off as an entry in the dropdown menu.  And it lets you browse into subdirectories in the file selector menu.

karoria, 2019-03-27:
Thanks @MitchBradley. You have given me my task. I will surely make some drawing kind of thing considering above points and submit to you. I am happy to help you. BTW, how can i upload those drawings to here? I am not well familier with git.

MitchBradley, 2019-03-27:
The comment box that I am typing in says, down at the bottom: "Attach files by dragging & dropping, selecting or pasting them."

karoria, 2019-03-27:
Ok. Now I got it. Previously I was typing through my mobile and somehow this option was not there. Thanks.

karoria, 2019-03-30:
Hi @MitchBradley. Plz find attached suggested layout as per my thoughts. Your comments please. Any more inputs required from my side, will be happy to help.
[Shopfloor Pendant Suggested Layout.pdf](https://github.com/cncjs/cncjs/files/3025440/Shopfloor.Pendant.Suggested.Layout.pdf)

spankurmonkey, 2019-03-30:
@karoria nice suggestion layout! +1 for that enhancement

karoria, 2019-04-01:
Waiting for your expert comments @MitchBradley.

MitchBradley, 2019-04-01:
I thought I had already commented but maybe I forgot to send them.  I like the approach but I think some of the buttons at the top - GoX0 etc - are too small.  Mis-hitting one of them can have bad consequences.

I think the welcome screen is unnecessary.  When you turn on your machine you are thinking about your task and anything that slows you down is irritating.  You just want to get right to it.

One approach to decluttering that top area is to have a separate screen for "set numeric value".  The buttons like "Go X" and "X=", that operate on the entered value, could be integrated into the pop-up calculator screen. or when you exit that calculator screen, there could be a following screen that shows the calculated value and then offers  you choices about what to do with it.

Another idea is to have a separate screen for MDI.  I often set up one-off jobs as a few MDI lines, so it would be helpful to have four or five lines instead of the two I have now.

karoria, 2019-04-01:
I agree with you. But i believe setting buttons of x y and z like x=0 etc should be on the jog screen only. I could not make it much larger size because of constrained space. However, I kept in mind 7 inch touch display for this. But i think if we design as per 5 inch screen then it would be great for mobile users. This demands more larger buttons and clutter free layout.

karoria, 2019-04-01:
Thought behind welcome screen is for homing purpose so that one cannot forget to home before starting work. However he can skip it also as per requirement. But yes, not adding much charm as you said. So we can remove it also. Tabs in right vertical column are arranged for better ergonomics as siemens and fanuc does with hard buttons.