cncjs/cncjs #405: Enhancement Request, mini keyboards and milling time

LukeBonathan, 2019-01-14:
#### Description

I love CNCJS, It's by far the most well designed and functional G code sender I've come across. I have been using some of the excellent macros posted here:

https://github.com/neilferreri/CNCjs-Macros

That said I have two enhancements I'd like to see in CNCJS. With my laptop I find it very hard to jog the machine around. I'd love to see native support for a mini keyboard similar to this?

https://www.amazon.co.uk/Wireless-Multi-media-Keyboard-Raspberry-Layout-Black-Black/dp/B00DUUF250/ref=pd_lpo_vtph_tr_t_2?_encoding=UTF8&psc=1&refRID=SVYMZ2VA00SYMZYNVCRZ

Ideally jogging and step increment.

I'd also love to see remaining time on a mulling cycle. 

#### Versions

- CNCjs: 1.9.6
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ x] Windows
- [ x] Mac
- [ ] Linux

nickm324, 2019-01-14:
you can already use a mini keyboard or keypad. 
[image]

TobyGWilliams, 2019-01-14:
You can find the estimated remaining time in the gcode widget here:

[image]

closed 2019-01-29 by cheton