cncjs/cncjs #406: Is CNCjs desktop app also a server for remote access?

leitenky, 2019-01-28:
#### Description

I have an old 24" monitor that I would like to use with CNCJs.  Currently I am running a pi 3b+ in kiosk mode with CNCJs server on it.  I find the performance unsatisfactory (as was expected). 
I was wondering if running Windows with CNCJs desktop app has a remote access (like the pi server) that I can check the camera/progress when needed (eg I can log in from my phone or another laptop to see the status of my current program)
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

langwadt, 2019-01-28:
teamviewer?

cheton, 2019-01-29:
@leitenky 

When running the desktop app, you can click **Home** under the **View** menu item, it will open a new browser window that contains the IP address and port, then you will know where to connect to the server.

[image]

leitenky, 2019-01-30:
Ok thanks for that.  For some reason I cannot log in from another PC/device.  I opened the firewall for the app and tried disabling antivirus. Any suggestions?

cheton, 2019-01-31:
If you're using public IP address rather than private IP address (i.e. `10.0.0.0/24`, `172.16.0.0/20`, `192.168.0.0/16`) within your network, you can run `cncjs --allow-remote-access` when launching the CNCjs server.

For the desktop app, you need to add a double-dash (`--`) before the options in the **Target** field, like below:

```
"C:\Users\Cheton Wu\AppData\Local\Programs\CNCjs\CNCjs.exe" -- --allow-remote-access
```

[image]

leitenky, 2019-02-01:
I tried the shortcut command but I still cannot access cnc.js from another machine. 
When I use the home command from the View tab the IP address it displays appears to be an external IP not an internal one. Is there a config file that can be edited for these things?

Thanks for all the help!

AustinSaintAubin, 2019-02-02:
@leitenky, recommend installing (or using a prebuilt image) CNCjs on the Raspberry Pi. The raspberry pi is generally more predictable and reliable than a windows desktop, mainly because of windows power saver, windows firewall, windows update... ext, all of which can have different configurations out of the box.
The slow performance you experienced on the Pi is almost entirely related to the UI.
If you connect to the UI from an other computer (connect to CNCjs from one computer, while it’s running on another) you will notice the computer that loads the UI will be working the hardest.

In summary, install CNCjs on the Pi, connect to the Pi from another computer. You will likly have the best experience his way.
As a side note: it’s likly windows firewall that is blocking you from accessing CNCjs from another computer, you can try disabling windows firewall and see if you can connect. But I still recommend CNCjs run in PI, accessed from other computer.

leitenky, 2019-02-02:
@austinsaintaubin thanks for the info.  I actually have reimaged my pi and it seems to be running much better (I must have tinkered with it to much before). I'm just trying to find the best way to get keyboard, mouse and display now in my garage.  I might see if I can dig up am old Android box and use a browser on it to give me the controls I'm looking for while in the shop ( I prefer to not use the phone to set it up the machine before a job)

Again thanks to everyone for the support and ideas.  I really like the customization available in CNC.JS!

MitchBradley, 2019-02-02:
In my workshop, I use a $100 Android tablet that talks wirelessly to the Pi. It uses the cncjs-shopfloor-tablet UI, which is optimized for touchscreens and avoids a lot of cncjs features that can be dangerous for production use (because important things like feedhold are tiny/hard to hit or even off-screen). For the infrequent keyboard use, I use the on-screen Android keyboard or, even more rarely, a wireless bluetooth keyboard.  With the touchscreen I don't need a mouse at all.  This results in a very clean, uncluttered, immune to dust shop setup.
Since I'm not running the UI on the Pi, a Pi Zero W with no display works fine.

leitenky, 2019-02-02:
Yes I've looked into that but my CNC is powered by Marlin so shopfloor tablet does not support Marlin unfortunately

MitchBradley, 2019-02-02:
If I get a minute I suppose I can try to add Marlin support.  I don't have a great way to test it, other than by turning on Marlin compatibility mode in g2core. If you are interested, perhaps you could do the beta testing.

neilferreri, 2019-02-02:
I was excited to see that this could be done with the desktop app, but I haven't been able to connect either. When I click the "home" option in the GUI, it opens a local machine IP only (127.0.0.1:xxxxx). 
How would I make it accessible on the local network?
I know I can use a Pi, but if this is possible through the Windows desktop app, it could be pretty valuable in at least one setup I have.

leitenky, 2019-02-02:
@MitchBradley I would be willing to give it a try and test it out for sure! I use the Mostly Printed CNC from v1 Engineering. If you get something up and running I could see if anyone on their forms is interested and testing also

neilferreri, 2019-02-05:
I seemed to have chimed in on a conversation that went another direction.  @leitenky were you ever able to connect from another PC. @cheton How do I get CNCjs to open an IP on my network rather than just on the machine?

leitenky, 2019-02-05:
@neilferreri I couldn't get it to work so I ended up using an old Android TV box and a raspberry pi.  It seems to be running pretty good that way so far

amrithmmh, 2019-02-09:
sorry for posting this question here,
this is a related question is it possible to send commands directly by connecting to ip port of cncjs (installed in rpi 3b + arduino grbl or smoothieboard) and send commands only like G28 or motion of X Y Z axis ?

AustinSaintAubin, 2019-02-09:
Look at also installing something like [Grid:Host -- multi-target file sender](https://github.com/GridSpace/grid-host).
You just have to disconnect the serial connection in CNCjs then this can connect.

closed 2019-04-20 by cheton

MitchBradley, 2020-01-07:
@cheton Do you know how to see console log output from the Windows electron app?  I'm trying to debug pendant usage in that app.  I have looked for log files in my Windows home directory and underneath %HOME%/AppData and can't find anything that looks like cncjs log output.

MitchBradley, 2020-01-07:
@cheton I was able to see log output by starting the electron app from a cmd prompt instead of by clicking the shortcut.

neilferreri, 2020-04-21:
Just an update for anyone finding this thread.  The solution can be found [in this post](https://github.com/cncjs/cncjs/issues/382#issuecomment-481218128)

MitchBradley, 2020-04-21:
And [here](https://github.com/cncjs/cncjs/wiki/Using-Pendants#using-the-app-with-browsers-on-other-machines-like-phones)