cncjs/cncjs #409: How can we help?

kikolobo, 2019-02-18:
Hi @cheton 

How can we help to make this project continue? I am not a great programmer but I am willing to donate $ to continue the cause!

It's going to be almost a year without an upgrade and this is by far the best CNC web app controller available.

Let me know how we can help!

cheton, 2019-02-18:
@kikolobo 

I'm making a big change in the [feature/redux](https://github.com/cncjs/cncjs/commits/feature/redux) branch, I'm planning to upgrade React to 16.8, remove several dependencies (e.g. Bootstrap), and merge new features.

However, my two little kids (3 yrs and 1.5yrs) almost occupy all my full time after work and also on weekends, so I may only have 2 hours per week for development recently.

I won't stop development on this project, but it may take me several months on newer update. I hope you will understand my situation and I really appreciate all your help (e.g. donations, improving documentations, making pull requests) to keep it moving forward. Thank you!

saxa, 2019-02-18:
@kikolobo I think the spirit of open source should be to develop whatever you need and make a PR. This would simply give @cheton the time to review and then merce into the current or table branch. If all would do it that way then would be a simple matter of announcing releases for @cheton something similira as you can see in the linux kernel.

kikolobo, 2019-02-18:
Totally understood! First thing first!

Thanks for the reply!

This is great software. You could have a commercial version!

MitchBradley, 2019-04-30:
The thing you could do that would help the most would be to improve the documentation.  Review wiki pages for clarity and completeness, etc.  If you have questions about how something works, ask, and then edit the answer into something coherent and put it in context on the wiki.

emcniece, 2022-03-26:
Hey @kikolobo, I know it's been a few years since you opened this ticket. I'm [getting started](https://github.com/cncjs/cncjs/discussions/749) on organizing some of the work here. If you're still interested, testing older issues to see if they're still valid would be of great help. You likely have much more experience than I do with this software, so maybe there's some [open issues](https://github.com/orgs/cncjs/projects/3/views/1?layout=table) that look easy to test and confirm.

No worries either way! Thank you for the offer 😄