cncjs/cncjs #411: CNCjs Marlin Windows 7

dzario, 2019-02-21:
Hi .
I download CNCjs for Windows , 
and test with my GeeeTech A10 with example code .
Work good .

But if use same electronic is possible Marlin firmware for real CNC ?
I test it, work X,Y,Z motion .. but is possible control any real pin on Mega2560 to control spindle or cooling ??

I design CNC where is possible use extruder and hotbed to mill or print .
but I need to control spindle , cooling , spindle speed etc.. 

regards.

AustinSaintAubin, 2019-02-22:
To answer your question, Yes.
Keep in mind that CNCjs just sends Gcode.

It’s not necessarily responsible for motion/control.  Your motion controller is responsible for or of the actual control like motion and pin control.

dzario, 2019-02-22:
ehm.. My GeeeTech board don't support spindle pin .. then I must this redefine on Marlin file .. 

thanks .

closed 2019-04-18 by cheton