cncjs/cncjs #412: webcam black in new chrome/firefox browser due to video tag'src deprecated for stream

dgntby, 2019-02-27:
#### Description

webcam black in new chrome/firefox browser due to video tag'src deprecated for stream

#### Versions

- CNCjs: latest
- Node.js: 8.11.3
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [x] souce run (preapare & start) 

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Windows
- [ ] Mac
- [ ] Linux
+++++++++++++++++++++++++++++++++++++++++++++++++++++
webcam,jsx:209
webcam,jsx:103
src in video element is depricated, srcObject should be used now.

I added a ref to video element and set the srcObject in  Webcam.mountedInstances.forEach to make it work again.

reference: https://stackoverflow.com/questions/48603402/how-to-set-srcobject-on-audio-element-with-react

ps. cncjs is awesome, thank you guys.

cheton, 2019-02-27:
Hi @dgntby 
Would you mind creating a pull request to resolve this issue? Thank you. 😃

dgntby, 2019-02-28:
@cheton 
I've thought about to resolve it but I'm not familiar with React and had concern of mass up your code, LOL.

anyway, I will  grasp basic of React knowledge and try:)

cheton, 2019-03-04:
You can check my pull request at https://github.com/cncjs/cncjs/pull/414

**Webcam.jsx**
https://github.com/cncjs/cncjs/blob/c40d822cacc1de0b92e6da35f0435a71e939f0ca/src/web/components/Webcam/Webcam.jsx

I will merge to the master branch if it works for you.

dgntby, 2019-03-05:
> @dgntby
> You can check my pull request at #414
> 
> **Webcam.jsx**
> https://github.com/cncjs/cncjs/blob/c40d822cacc1de0b92e6da35f0435a71e939f0ca/src/web/components/Webcam/Webcam.jsx
> 
> I will merge to the master branch if it works for you.

@cheton 

I intended to fix it by this solution, but failed. The compiler complained React.createRef in PureComponent. that's why I fixed it in a ugly way, LOL

I guess if you passed the compile and webpack process, this is a much elagant solution:)

cheton, 2019-03-05:
Ooops. That's my mistake. I fixed this issue with the commit https://github.com/cncjs/cncjs/commit/85a2f63e46ef0830a06fa24bea6d6c8b7a031bd5, which is inside the [feature/redux](https://github.com/cncjs/cncjs/tree/feature/redux) branch.

React.createRef() is only supported since React 16.3, I will make another change that is suitable for the master branch.

cheton, 2019-03-05:
@dgntby New changes were made in PR https://github.com/cncjs/cncjs/pull/414. It will make Webcam component compatible with React 15.6.

closed 2019-03-05 by cheton

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)

PR #414 (2019-03-05): Fixed a deprecation issue that `srcObject` should be used for a `MediaStream` object (#412)
Fixed #412