cncjs/cncjs #413: TinyWeb Console Install from Wiki - cannot complete

spankurmonkey, 2019-02-28:
#### Description

Cannot complete TinyWeb Console install per the linked Wiki - https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-TinyWeb-Console-for-320x240-LCD-Display

I get to this line:
```
# How-to Start CNCjs w/ mounted TinyWeb
cncjs -m /tinyweb:/home/pi/tinyweb
```
then the PI hangs in the console with the following
```
pi@cncjs:~ $ cncjs -m /tinyweb:/home/pi/tinyweb
2019-02-28T21:11:39.107Z - info: init Loading configuration from "/home/pi/.cncr                                                                                                      c"
2019-02-28T21:11:39.145Z - info: init Mounting a directory "/home/pi/tinyweb" to                                                                                                       serve requests starting with /tinyweb
2019-02-28T21:11:40.103Z - error: init { Error: listen EADDRINUSE: address alrea                                                                                                      dy in use 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (net.js:1277:14)
    at listenInCluster (net.js:1325:12)
    at doListen (net.js:1458:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:745:11)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:743:3)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }
Error: { Error: listen EADDRINUSE: address already in use 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (net.js:1277:14)
    at listenInCluster (net.js:1325:12)
    at doListen (net.js:1458:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)
    at Function.Module.runMain (internal/modules/cjs/loader.js:745:11)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:743:3)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }
```

If I reboot the PI the TFT is still blank nothing displayed.

#### Versions

- CNCjs: 1.9.15
- Node.js: latest followed the wiki step by step
- NPM: latest followed the wiki step by step

#### How Do You Install CNCjs?
Through PUTTY interface
- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-02-28:
This is a really common problem that bites everybody at some time.  The situation is that cncjs is already running, so when you run the cncjs command, it tries to listen on the web address/port and finds that somebody else already "owns" that port - namely the already-running instance of cncjs.
The solution is to kill the old cncjs process first.
One way to do that is to type
`killall node`
However, depending on which instructions that you used to set up your Pi for cncjs, it is possible that something like pm2 might notice that cncjs stopped and automatically restart it.  If so you have to tell that thing not to do it - or else change that things configuration so it restarts cncjs with the arguments that you want - namely the "-m tinyweb:/home/pi/tinyweb" arguments.
Personally, I think that using pm2 on a Pi is a bad idea.  pm2 is a heavyweight application with a lot  of bells and whistles, when all you really want is to autostart cncjs on boot up. Linux already has numerous ways to do that, so adding yet another one, and especially another one that carries all the overhead of node.js, is not the right approach for a resource-limited machine like a Pi.

spankurmonkey, 2019-02-28:
I wish I knew more but do not so I am not sure what I should do next to try to the lCD running.

MitchBradley, 2019-02-28:
Are you using pm2?

spankurmonkey, 2019-02-28:
Hi Mitch - I am a novice and merely followed all of the steps in the wiki to get the CNCjs running on the PI.
I can connect to the PI/CNC js on my local IP.
Once I got that working and tested it with my machine I moved on to the TinyWeb Console  -- this is where I am stuck .. I got the tft today https://www.amazon.com/gp/product/B00OUQHJK4 
I shut down the PI through SSH and then installed the TFT and booted it back up but the screen is lit but blank... I tried the steps again in the Wiki and this is where I landed as I am not sure how to make it work or what I did wrong, etc.

MitchBradley, 2019-02-28:
There are several wiki pages relating to Pi, but I will guess that you mean the "Raspberry Pi Setup Guide" and its subpages.  The instructions in the "Update CNCjs" and "Update Production Process Manager [PM2]" sections require you to type a lot of commands that begin with "pm2".  That is what I meant by "are you using pm2".

If you did do that, then we have to turn off pm2 because it is going to keep getting in you way when you try to run the lcd pendant.  Here's what you should try:

```
pm2 stop cnc
cncjs -m /tinyweb:/home/pi/tinyweb

```

spankurmonkey, 2019-02-28:
Mitch - I'm wondering if I got ahead of myself.... I just noticed on the Amazon detials for the TFT that I have to follow a setup guide that installs and configures the screen to work.   

I am guessing I should have done that first.....

spankurmonkey, 2019-02-28:
Well what ever I just did trying to follow the TFT setup just killed the image on the Pi.. will not even boot back up.

So for a novice what would you suggest .... I am going need to reinstall so should I be installing just noobs/raspibain and then set up the TFT and then go down the CNCjs install rabbit hole?

MitchBradley, 2019-02-28:
That looks similar to the Pi TFT that I got from Adafruit.  Getting it to work was a lot of trouble - and I have being doing this kind of thing for 40+ years.

I expect that the step that "killed" your Pi was the bootloader thing.

You are in for a lot of wrangling.  The main problem is that any of the steps could be a little bit wrong due to changes between system versions.

Get your system working first then worry about CNCjs.  And when you do get back to CNCjs, do not do any of the steps involving pm2.  I will tell you a different way when the time comes.

MitchBradley, 2019-02-28:
When you say "will not boot back up", do you mean that it will not run the window system, or is it so bad that you don't get the text messages on the HDMI monitor?

spankurmonkey, 2019-02-28:
Yes - nothing on the HDMI output... screen blank - Wifi wouldn't come back up, etc.

I have reformatted the sd card, using PiBakery got it all set with the noobs and raspbian... I can now see the desktop environment through the HDMI output.

I am guessing now I should mess with trying to get the TFT working.. if I do then come back and ask about CNCjs install correct?

MitchBradley, 2019-02-28:
Let me have a look at those patches before you install them.  I'm concerned that the "bootloader" one might be what caused your system not to boot.

spankurmonkey, 2019-02-28:
The patches in this guide you mean?
https://www.adrive.com/public/9zUKSu/Raspberry%20Pi%202.8TFT.pdf

MitchBradley, 2019-02-28:
Yes

spankurmonkey, 2019-02-28:
ok. will wait to hear back from you

MitchBradley, 2019-02-28:
That patch file "raspberrypi-bootloader-adafruit-20140724-1.deb" looks scary.  It replaces the entire Linux kernel on your SD card.  That could go wrong in so many ways.  Let me look through the adafruit docs to see if there is a better approach.

spankurmonkey, 2019-02-28:
Perhaps I should look for a different TFT a better supported one ?

MitchBradley, 2019-02-28:
https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install-2

spankurmonkey, 2019-02-28:
It may be better to use a trusted TFT .... lesson learned.  thanks for your help.

MitchBradley, 2019-02-28:
I think the problem is not with your TFT card, but rather with the simple-minded instructions referenced on the Amazon page.  Those instructions work only in a very specific environment.

spankurmonkey, 2019-02-28:
So do you think it's worth figuring it out... ?  I wouldn't know how too.

MitchBradley, 2019-02-28:
The TFT hardware is pretty much generic.  There is essentially only one chip that does the work, and every board uses the same chip.  The problem is the software, in particular the fact that Linux/Raspbian changes about 4 times per year.  So if you try to make a snapshot of an image that supports the TFT, it is pretty much guaranteed to be obsolete in a heartbeat.
The instructions on the adafruit page that I referenced above look good.  There is an installer script that works out what to do on the fly.  I think that, if you follow those instructions, you have a good chance of winning.  Use the March 13, 2018 Raspbian as recommended.
Select choice #1 because it is a 2.8" resistive screen.
Use the PiTFT as HDMI Mirror (Best for Raspbian Full/PIXEL) option.

spankurmonkey, 2019-02-28:
Ok. will give it a try.

MitchBradley, 2019-02-28:
I edited the TinyWeb instructions on the wiki, so old instances of cncjs are stopped before running cncjs/TinyWeb.

spankurmonkey, 2019-02-28:
Its asking me if I want the console to appear on the PiTFT display...

spankurmonkey, 2019-02-28:
Is the TinyWeb when working only limited to the 2.8 TFT or can it be used on any PI TFT display ?

spankurmonkey, 2019-02-28:
The TFT now displays the desktop but noticed the touch feature does not work. I pulled the protective cover off used my finger and a stylus and nothing....

MitchBradley, 2019-03-01:
I can work with you tomorrow on this.  I have to do some other stuff for the next few hours.

MitchBradley, 2019-03-01:
Go through the steps on this page to check touchscreen function:

https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/resistive-touchscreen-manual-install-calibrate

Yes I know it should have happened automagically but obviously something didn't take.

spankurmonkey, 2019-03-01:
Got it--- I am trying those steps but the step that says 
```
Remove and re-install the touchscreen with

sudo rmmod stmpe_ts; sudo modprobe stmpe_ts
Then type ls -l /dev/input/touchscreen
```

I am getting this response 

```
pi@cncjs:~ $ sudo rmmod stmpe_ts; sudo modprobe stmpe_ts
pi@cncjs:~ $ ls -l /dev/input/touchscreen
ls: cannot access '/dev/input/touchscreen': No such file or directory

```

spankurmonkey, 2019-03-01:
I have tried these steps a few times and sometimes I get this error too:

```
pi@cncjs:~ $ sudo nano /etc/udev/rules.d/95-stmpe.rules
pi@cncjs:~ $ sudo rmmod stmpe_ts; sudo modprobe stmpe_ts
rmmod: ERROR: Module stmpe_ts is not currently loaded
```

spankurmonkey, 2019-03-01:
I have confirmed this step

```
sudo nano /etc/udev/rules.d/95-stmpe.rules
SUBSYSTEM=="input", ATTRS{name}=="stmpe-ts", ENV{DEVNAME}=="*event*", SYMLINK+="input/touchscreen" 
```
Even after a reboot this rule is still present when I run

`sudo nano /etc/udev/rules.d/95-stmpe.rules`

spankurmonkey, 2019-03-01:
Well I am throwing in the towel on this and returning the screen ... it's seems to be more effort than it is worth.

MitchBradley, 2019-03-01:
It does appear that the touchscreen hardware is no working.

MitchBradley, 2019-03-01:
I'll close this issue for now.

closed 2019-03-01 by MitchBradley