cncjs/cncjs #415: CNCjs Shop-floor-tablet

mcmillanrs, 2019-03-04:
#### Description

I started cncjs with the shopfloor-tablet mounting point.  The full UI works great but when I try to load the tablet at http://ip address:8000/tablet/ in Safari or Chromium the interface loads with a Trying
Controller Type list box with GRBL and empty Port listbox and a Baud rate list box of 115200.  The Port listbox does not populate and one cannot type the port into it.  I created a .cncrc with the port which works on the full UI but this makes no difference.  I hope you can help me.  Cheers

#### Versions

- CNCjs: 1.9.15
- Node.js: 10.15.1
- NPM: 6.8.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2019-03-05:
Get this version
https://github.com/cncjs/cncjs-shopfloor-tablet/tree/MitchBradley-patch-all-serial-ports
The master version is tweaked to look only for the g2core controller that I have.

mcmillanrs, 2019-03-05:
Hello Mitch. Thank you for that. Cheers Rod

iPad

closed 2019-03-06 by MitchBradley

mcmillanrs, 2019-03-06:
Thank you Mitch: everything works perfectly now. Cheers Rod

iPad