cncjs/cncjs #416: Axis Widget does not remember configured axis selections

bradanlane, 2019-03-05:
#### Description

Running the pre-built Raspberry Pi 1.0.1 image on RPi3B. I ma using CNCjs for a laser engraver. The machine has X&Y (no Z axis). Used the Axis Widget configuration icon to get to the settings for the widget. Deselected "Z". Saved changes. Returning to the widget the Z axis has been removed from teh Axis table. Exit the browser and open again to CNCjs. The Z-axis is displayed again.

I looked at the workspace file and there does not appear to be settings for Axis are being saved.

#### Versions

- CNCjs: 1.9.15
- Node.js: 10.10.0
- NPM: 6.4.1

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- [X] pre-built Raspberry Pi image

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [X] Linux

cheton, 2019-03-05:
I can confirm it is a bug in 1.9.15, the axes settings will not be saved if Y/Z axis is not selected. I will fix it in 1.9.16.

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)

closed 2019-03-07 by cheton