cncjs/cncjs #420: modal dialog boxes do not have focus

bradanlane, 2019-03-10:
#### Description

Clicking on the "three dots" of a widget and then clicking remove. Neither button [cancel / OK] has focus and hitting the "TAB" key shows that the underlying app still have keyboard focus.

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [ ] Download the CNCjs Desktop Application
- Raspberry Pi pre-built image

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2019-03-13:
Could you explain in more details? Are you unable to click UI buttons with a mouse?

bradanlane, 2019-03-13:
Clicking with the muse shifts UI focus to the modal dialog box. However, without first clicking the dialog box, the focus is still on the underlying UI. The focus should programatically switch to the modal dialog box rather than require the mouse click first. It’s actually the definition of “modal”.