cncjs/cncjs #423: Windows desktop build EXE fails to run properly

zachallaun, 2019-03-11:
#### Description

I'm making some minor changes to the current version of cncjs based on my personal preferences: https://github.com/cncjs/cncjs/compare/master...zachallaun:ch-ch-changes?expand=1

Dev is happening on my Mac and I'm building the Win64 version for use on the PC connected to my Grbl controller. I'm building the EXE and then transferring it to the PC with Dropbox. The build seems to succeed and I get no errors when installing cncjs on the PC, but when I try to open the application after the install, nothing happens. Well, it _seems_ like nothing happens, but apparently _something_ is happening because the installer complains that cncjs is already running if I try to reinstall or if I reinstall the original version.

I'm building the Win64 version with:

```sh
npm run prepare && npm run build:win-x64
```

I was able successfully build, install, and use the Mac desktop version with the following:

```sh
npm run prepare && npm run build:mac-x64
```

Any guidance would be much appreciated.

#### Versions

- CNCjs: 1.9.16 (fork off of master)
- Node.js: 11.9.0
- NPM: 6.9.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [x] Mac
- [ ] Linux

cheton, 2019-03-20:
Hi @zachallaun 

From my past experiences, you can build only mac-x64/linux-x64 versions on a Mac. For building Windows desktop, you have to `npm run prepare && npm run build:win-x64` **in a Windows x64 environment (w/ Visual Studio installed)** to get the expected result. I don't know the actual reason why it won't work, but I believe that it would require Visual Studio to build native binaries.

zachallaun, 2019-03-20:
Hi @cheton. Thanks for the response! In the meantime, I've *sort of* gotten things working by running the code in Windows Subsystem for Linux and connecting from a browser, but it seems to only be able to connect to the grbl controller about half the time (I haven't been able to figure out what keeps it from connecting sometimes).

I'll give compiling w/ Visual Studio a shot. Going to close this issue for now. Thanks for your help!

closed 2019-03-20 by zachallaun